TheSpaghettiDetective/obico-server #334: SMTP timeout - cannot send mail

Axolord, 2020-12-19:
Hi, probably I am just to dump to set it up, but the sending of mails does not work for me.
I use a hosted mail server at netcup.net and configured the SMTP settings just like in thunderbird and also tried gmail with no success.

I am behind a traefik reverse proxy and have the following docker-compose config:
```
version: '2.4'

x-web-defaults: &web-defaults
  restart: unless-stopped
  build:
    context: web
    dockerfile: "Dockerfile${DOCKERFILE_VERSION}"
  volumes:
    - ./web:/app
  depends_on:
    - redis
  dns:
    - 1.1.1.1
    - 1.0.0.1
  environment:
    EMAIL_HOST: 'mx2e33.netcup.net'    # -> such as 'smtp.gmail.com'
    EMAIL_HOST_USER: 'user@example.com'  # -> such as your email address for a Gmail account
    EMAIL_HOST_PASSWORD: 'mypassword'   # -> your email account password
    EMAIL_PORT: 465 # DO NOT surround it with quotes. Otherwise email won't be sent!
    EMAIL_USE_TLS: 'True'
    DEFAULT_FROM_EMAIL: 'user@example.com'
    DEBUG: 'False'    # Don't set DEBUG to True, otherwise the static files will be cached in browser until hard-refresh
    SITE_USES_HTTPS: 'True' # set it to 'True' if https is set up
    SOCIAL_LOGIN: 'False'
    REDIS_URL: 'redis://redis:6379'
    DATABASE_URL: 'sqlite:////app/db.sqlite3'
    INTERNAL_MEDIA_HOST: 'http://web:3334'
    ML_API_HOST: 'http://ml_api:3333'
    ACCOUNT_ALLOW_SIGN_UP: 'False'  # -> set to 'True' if you want to open sign up form
    WEBPACK_LOADER_ENABLED:

    #### Optional env vars below this line ###
    #TWILIO_ACCOUNT_SID: https://django-twilio.readthedocs.io/en/latest/settings.html for how to find and set the TWILIO_XXX vars
    #TWILIO_AUTH_TOKEN:
    #TWILIO_FROM_NUMBER:
    #ENALBE_HISTORY:
    #ML_API_TOKEN:
    #SENTRY_DSN:
    TELEGRAM_BOT_TOKEN: 'token' # https://core.telegram.org/bots#6-botfather for how to create a telegram bot and get its token

services:
  ml_api:
    hostname: ml_api
    restart: unless-stopped
    dns:
      - 1.1.1.1
      - 1.0.0.1
    build:
      context: ml_api
    environment:
        DEBUG: 'True'
        FLASK_APP: 'server.py'
        #ML_API_TOKEN:
        #HAS_GPU: 'False'
    command: bash -c "gunicorn --bind 0.0.0.0:3333 --workers 1 wsgi"

  web:
    <<: *web-defaults
    hostname: web
    ports:
      - "3334:3334"
    dns:
      - 1.1.1.1
      - 1.0.0.1
    labels:
      - "traefik.enable=true"
      - "traefik.port=3334"
      - "traefik.http.services.spaghetti.loadbalancer.server.port=3334"
      - "traefik.http.routers.spaghetti.middlewares=spaghetti-redirect-websecure"
      - "traefik.http.routers.spaghetti.rule=Host(`spaghetti.mydomain.com`)"
      - "traefik.http.routers.spaghetti.entrypoints=web"
      - "traefik.http.routers.spaghetti-secure.entrypoints=websecure"
      - "traefik.http.routers.spaghetti-secure.rule=Host(`spaghetti.mydomain.com`)"
      - "traefik.http.routers.spaghetti-secure.tls.certresolver=mytlschallenge"
      - "traefik.http.middlewares.spaghetti-redirect-websecure.redirectscheme.scheme=https"
    networks:
      - traefik_proxy
    depends_on:
      - ml_api
    command: sh -c "python manage.py collectstatic --noinput && python manage.py migrate && python manage.py runserver --noreload 0.0.0.0:3334"

  tasks:
    <<: *web-defaults
    hostname: tasks
    dns:
      - 1.1.1.1
      - 1.0.0.1
    command: sh -c "celery -A config worker --beat -l info -c 2 -Q realtime,celery"

  redis:
    restart: unless-stopped
    image: redis:5.0-alpine

networks:
  traefik_proxy:
    external:
      name: traefik_proxy
```

If I try to send the test mail the site just loads indefinitely until I reload it, then the logs state:
```
web_1     | backoff      INFO     Backing off send_email(...) for 0.7s  smtplib.SMTPServerDisconnected: Connection unexpectedly closed)
web_1     | daphne.server WARNING  Application instance <Task pending coro=<AsgiHandler.__call__() running at /usr/local/lib/python3.7/site-packages/channels/http.py:192> wait_for=<Future pending cb=[_chain_future.<locals>._call_check_cancel() at /usr/local/lib/python3.7/asyncio/futures.py:351, <TaskWakeupMethWrapper object at 0x7faf33b4add0>()]>> for connection <WebRequest at 0x7faf33b29fd0 method=POST uri=/admin/app/user/ clientproto=HTTP/1.1> took too long to shut down and was killed.
```

The telegram notification works fine though.
Any help is appreciated, thanks!

kennethjiang, 2020-12-19:
This looks like an environment or configuration issue. Not something we can change code to fix. Your best bet is probably TSD's discord server: https://discord.gg/NcZkQfj

Axolord, 2020-12-19:
Will ask there for help, thanks!

closed 2020-12-19 by Axolord