TheSpaghettiDetective/obico-server #335: docker Compose failed "ml_api failed to build"

BasharNchtn, 2020-12-19:
Hey, I tried to install the server on a fedora system, but I have this error :
`$ sudo docker-compose up -d
WARNING: The DOCKERFILE_VERSION variable is not set. Defaulting to a blank string.
Building ml_api
Step 1/7 : FROM thespaghettidetective/ml_api:base
 ---> 6eaa0374427c
Step 2/7 : WORKDIR /app
 ---> Using cache
 ---> 7e0b6fdf41a7
Step 3/7 : EXPOSE 3333
 ---> Using cache
 ---> a67d19259d75
Step 4/7 : ADD . /app
 ---> Using cache
 ---> c44678fa02c6
Step 5/7 : RUN pip install --upgrade pip
 ---> Running in 70e52bcc5867
ERROR: Service 'ml_api' failed to build: OCI runtime create failed: this version of runc doesn't work on cgroups v2: unknown`

For information docker version is  19.03.11 and docker-compose 1.25.4

kennethjiang, 2020-12-20:
What is the hardware and host OS you are running?

BasharNchtn, 2020-12-21:
The hardware is a desktop computer not very recent but quite powerful (core i5 3.1 GHz for the CPU with 32 Go RAM), and OS is linux Fedora 32.

kennethjiang, 2020-12-23:
I started a Fedora 32 64 bits VM in the cloud but couldn't reproduce the problem:

```
[fedora@ip-172-31-79-218 TheSpaghettiDetective]$ sudo docker-compose up -d
WARNING: The DOCKERFILE_VERSION variable is not set. Defaulting to a blank string.
Creating network "thespaghettidetective_default" with the default driver
Building ml_api
Step 1/7 : FROM thespaghettidetective/ml_api:base
base: Pulling from thespaghettidetective/ml_api
34667c7e4631: Pull complete
d18d76a881a4: Pull complete
119c7358fbfc: Pull complete
2aaf13f3eff0: Pull complete
4d96b2dafaa5: Pull complete
f8c41b380cab: Pull complete
d2c1b4858446: Pull complete
ba985066a7db: Pull complete
60930c6a628c: Pull complete
efe5c8ec620f: Pull complete
a254c1b788ef: Pull complete
c59301333085: Pull complete
342e31f8eddc: Pull complete
Digest: sha256:53e2ffcf47a0ee4f0c65ff3385cba28093053e36aa12c26b4afd27ded1af3547
Status: Downloaded newer image for thespaghettidetective/ml_api:base
 ---> 6eaa0374427c
Step 2/7 : WORKDIR /app
 ---> Running in 4e24fab6e3d0
Removing intermediate container 4e24fab6e3d0
 ---> 0126257dbedd
Step 3/7 : EXPOSE 3333
 ---> Running in c041155ea669
Removing intermediate container c041155ea669
 ---> ecd9ef6469aa
Step 4/7 : ADD . /app
 ---> a9aad0605e8e
Step 5/7 : RUN pip install --upgrade pip
 ---> Running in 08180f040a0a
Collecting pip
  Downloading https://files.pythonhosted.org/packages/54/eb/4a3642e971f404d69d4f6fa3885559d67562801b99d7592487f1ecc4e017/pip-20.3.3-py2.py3-none-any.whl (1.5MB)
Installing collected packages: pip
  Found existing installation: pip 19.0.3
    Uninstalling pip-19.0.3:
      Successfully uninstalled pip-19.0.3
Successfully installed pip-20.3.3
Removing intermediate container 08180f040a0a
 ---> 6360f0d77c25
Step 6/7 : RUN pip install -r requirements.txt
 ---> Running in dd8b993ffbae
DEPRECATION: Python 3.5 reached the end of its life on September 13th, 2020. Please upgrade your Python as Python 3.5 is no longer maintained. pip 21.0 will drop support for Python 3.5 in January 2021. pip 21.0 will remove support for this functionality.
Requirement already satisfied: ipdb in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 1)) (0.12)
Requirement already satisfied: flask>=1.0 in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 2)) (1.0.2)
Requirement already satisfied: redis==3.0.1 in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 3)) (3.0.1)
Requirement already satisfied: newrelic==4.12.0.113 in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 4)) (4.12.0.113)
Requirement already satisfied: raven[flask]==6.10.0 in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 5)) (6.10.0)
Requirement already satisfied: requests==2.21.0 in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 6)) (2.21.0)
Requirement already satisfied: gunicorn==19.9.0 in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 7)) (19.9.0)
Requirement already satisfied: blinker>=1.1 in /usr/local/lib/python3.5/dist-packages (from raven[flask]==6.10.0->-r requirements.txt (line 5)) (1.4)
Requirement already satisfied: urllib3<1.25,>=1.21.1 in /usr/local/lib/python3.5/dist-packages (from requests==2.21.0->-r requirements.txt (line 6)) (1.24.2)
Requirement already satisfied: idna<2.9,>=2.5 in /usr/local/lib/python3.5/dist-packages (from requests==2.21.0->-r requirements.txt (line 6)) (2.8)
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in /usr/local/lib/python3.5/dist-packages (from requests==2.21.0->-r requirements.txt (line 6)) (3.0.4)
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python3.5/dist-packages (from requests==2.21.0->-r requirements.txt (line 6)) (2019.3.9)
Requirement already satisfied: Jinja2>=2.10 in /usr/local/lib/python3.5/dist-packages (from flask>=1.0->-r requirements.txt (line 2)) (2.10.1)
Requirement already satisfied: click>=5.1 in /usr/local/lib/python3.5/dist-packages (from flask>=1.0->-r requirements.txt (line 2)) (7.0)
Requirement already satisfied: Werkzeug>=0.14 in /usr/local/lib/python3.5/dist-packages (from flask>=1.0->-r requirements.txt (line 2)) (0.15.2)
Requirement already satisfied: itsdangerous>=0.24 in /usr/local/lib/python3.5/dist-packages (from flask>=1.0->-r requirements.txt (line 2)) (1.1.0)
Requirement already satisfied: MarkupSafe>=0.23 in /usr/local/lib/python3.5/dist-packages (from Jinja2>=2.10->flask>=1.0->-r requirements.txt (line 2)) (1.1.1)
Requirement already satisfied: setuptools in /usr/lib/python3/dist-packages (from ipdb->-r requirements.txt (line 1)) (20.7.0)
Requirement already satisfied: ipython>=5.1.0 in /usr/local/lib/python3.5/dist-packages (from ipdb->-r requirements.txt (line 1)) (7.4.0)
Requirement already satisfied: decorator in /usr/local/lib/python3.5/dist-packages (from ipython>=5.1.0->ipdb->-r requirements.txt (line 1)) (4.4.0)
Requirement already satisfied: traitlets>=4.2 in /usr/local/lib/python3.5/dist-packages (from ipython>=5.1.0->ipdb->-r requirements.txt (line 1)) (4.3.2)
[... 209 lines cut ...]
 ---> ad1bee20bd2a
Step 3/7 : EXPOSE 3334
 ---> Using cache
 ---> 7db88db3210f
Step 4/7 : ADD . /app
 ---> Using cache
 ---> e2370ccdbbbf
Step 5/7 : RUN pip install -U pip
 ---> Using cache
 ---> 2588828f4827
Step 6/7 : RUN pip install -r requirements.txt
 ---> Using cache
 ---> 600c9794ca89
Step 7/7 : RUN python manage.py collectstatic --noinput -c
 ---> Using cache
 ---> f43885504f40

Successfully built f43885504f40
Successfully tagged thespaghettidetective_tasks:latest
WARNING: Image for service tasks was built because it did not already exist. To rebuild this image you must use `docker-compose build` or `docker-compose up --build`.
Creating thespaghettidetective_redis_1  ... done
Creating thespaghettidetective_ml_api_1 ... done
Creating thespaghettidetective_tasks_1  ... done
Creating thespaghettidetective_web_1    ... done
[fedora@ip-172-31-79-218 TheSpaghettiDetective]$

BasharNchtn, 2020-12-26:
Strange, I still have the problem and can not understand why you couldn't reproduce it.
I see in your log that you use python 3.5 whereas my version is 3.8.6. (But in your case you have a Warning about that, so it will be a shame of newer version make install not working).
I try manually to do "sudo python -m pip install --upgrade pip", as written in log for step 5/7 (which seems fail on my computer)
But it works fine. 
I don't know well docker (and docker-compose) so I really don't understand what's happened here.

I check that my fedora is up to date, turning on last official kernel (actually 5.9.15 )

I have no idea of what "Service 'ml_api' failed to build: OCI runtime create failed: this version of runc doesn't work on cgroups v2: unknown" mean and how to fix that to continue the install process

So can you help me ? It's a pity if I couldn't use TSD just because a python install seem not working

kennethjiang, 2020-12-26:
Unfortunately I'm not sure how I can help you without the ability to reproduce this problem... :(

BasharNchtn, 2020-12-27:
Finally I search a bit more on the web and find [here](https://bugzilla.redhat.com/show_bug.cgi?id=1746355) that there is some problem with current docker version and fedora kernel. A solution was to have more recent version of docker that I can install following [the main documentation](https://docs.docker.com/engine/install/fedora/) instead of the version from Fedora official repository. With that it seems to work and i have the same log than you. I suppose so that you have this version of docker in your test.

By the way it solves the problem and issue can be closed, thanks for your help. I suggest to improve your software just to add in the main documentation for docker installation to follow the link in linux (instead of only Ubuntu).

kennethjiang, 2020-12-27:
Great to know you found the solution. I added the links to Fedora and CentOS as they have been tested our users. I'm not sure if there are a Docker link for all Linux distros but I might have missed something.

closed 2020-12-27 by kennethjiang