TheSpaghettiDetective/obico-server #339: Waiting For Connection

Zexxen-Zalos, 2020-12-22:
I was able to add my first printer on a server running on a jetson nano but when I tried to add a second printer the plugin in octoprint states the secret token is valid but the server never gets past the waiting for the spaghetti detective plugin to connect. I opened the log and I noticed it keeps repeating "HTTP GET /api/v1/printers/4/ 200 [0.04, 192.168.50.79:58048]" so I'm guessing it has something to do with that.

`System check identified no issues (0 silenced).
web_1     | December 21, 2020 - 01:14:40
web_1     | Django version 2.2.10, using settings 'config.settings'
web_1     | Starting ASGI/Channels version 2.4.0 development server at http://0.0.0.0:3334/
web_1     | Quit the server with CONTROL-C.
web_1     | daphne.server INFO     Killed 3 pending application instances
web_1     | django.channels.server INFO     WebSocket DISCONNECT /ws/janus/1/ [192.168.50.79:57960]
web_1     | django.channels.server INFO     WebSocket DISCONNECT /ws/dev/ [192.168.50.39:37666]
web_1     | django.channels.server INFO     WebSocket DISCONNECT /ws/web/1/ [192.168.50.79:58005]
web_1     |
web_1     | 0 static files copied to '/app/static_build', 257 unmodified, 545 post-processed.
web_1     | /usr/local/lib/python3.7/site-packages/daphne/server.py:15: UserWarning: Something has already installed a non-asyncio Twisted reactor. Attempting to uninstall it; you can fix this warning by importing daphne.server early in your codebase or finding the package that imports Twisted and importing it later on.
web_1     |   UserWarning,
web_1     | Operations to perform:
web_1     |   Apply all migrations: account, admin, app, auth, channels_presence, contenttypes, sessions, sites, socialaccount
web_1     | Running migrations:
web_1     |   No migrations to apply.
web_1     |   Your models have changes that are not yet reflected in a migration, and so won't be applied.
web_1     |   Run 'manage.py makemigrations' to make new migrations, and then re-run 'manage.py migrate' to apply them.
web_1     | /usr/local/lib/python3.7/site-packages/daphne/server.py:15: UserWarning: Something has already installed a non-asyncio Twisted reactor. Attempting to uninstall it; you can fix this warning by importing daphne.server early in your codebase or finding the package that imports Twisted and importing it later on.
web_1     |   UserWarning,
web_1     | /usr/local/lib/python3.7/site-packages/daphne/server.py:15: UserWarning: Something has already installed a non-asyncio Twisted reactor. Attempting to uninstall it; you can fix this warning by importing daphne.server early in your codebase or finding the package that imports Twisted and importing it later on.
web_1     |   UserWarning,
web_1     | daphne.server INFO     HTTP/2 support not enabled (install the http2 and tls Twisted extras)
web_1     | daphne.server INFO     Configuring endpoint tcp:port=3334:interface=0.0.0.0
web_1     | daphne.server INFO     Listening on TCP address 0.0.0.0:3334
web_1     | django.channels.server INFO     WebSocket HANDSHAKING /ws/dev/ [192.168.50.39:38448]
web_1     | newrelic.core.agent INFO     New Relic Python Agent (4.12.0.113)
web_1     | newrelic.core.agent WARNING  The Python Agent is not enabled.
web_1     | django.channels.server INFO     WebSocket CONNECT /ws/dev/ [192.168.50.39:38448]
web_1     | django.channels.server INFO     WebSocket HANDSHAKING /ws/web/1/ [192.168.50.79:58036]
web_1     | django.channels.server INFO     WebSocket CONNECT /ws/web/1/ [192.168.50.79:58036]
web_1     | django.channels.server INFO     HTTP GET /printers/ 200 [0.41, 192.168.50.79:58042]
web_1     | django.channels.server INFO     WebSocket DISCONNECT /ws/web/1/ [192.168.50.79:58036]
web_1     | api.consumers WARNING  WebConsumer: Closed websocket with code: None
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/?with_archived=true 200 [0.05, 192.168.50.79:58042]
web_1     | django.channels.server INFO     HTTP GET /static/frontend/img/octoprint-tunnel.png 304 [0.01, 192.168.50.79:58042]
web_1     | django.channels.server INFO     HTTP GET /static/img/favicon-32x32.4c5b80fc1e46.png 200 [0.02, 192.168.50.79:58045]
web_1     | django.channels.server INFO     HTTP GET /static/frontend/img/3d_printer.png 304 [0.00, 192.168.50.79:58047]
web_1     | django.channels.server INFO     WebSocket HANDSHAKING /ws/web/1/ [192.168.50.79:58046]
web_1     | django.channels.server INFO     HTTP GET /static/frontend/img/loading.gif 304 [0.01, 192.168.50.79:58048]
web_1     | django.channels.server INFO     WebSocket CONNECT /ws/web/1/ [192.168.50.79:58046]
web_1     | django.channels.server INFO     WebSocket HANDSHAKING /ws/janus/1/ [192.168.50.79:58049]
web_1     | django.channels.server INFO     WebSocket CONNECT /ws/janus/1/ [192.168.50.79:58049]
web_1     | django.channels.server INFO     WebSocket DISCONNECT /ws/janus/1/ [192.168.50.79:58049]
web_1     | api.consumers WARNING  WebConsumer: Closed websocket with code: None
web_1     | django.channels.server INFO     HTTP GET /printers/new/ 200 [0.14, 192.168.50.79:58048]
web_1     | django.channels.server INFO     WebSocket DISCONNECT /ws/web/1/ [192.168.50.79:58046]
web_1     | api.consumers WARNING  WebConsumer: Closed websocket with code: None
web_1     | django.channels.server INFO     HTTP POST /printers/new/ 302 [0.12, 192.168.50.79:58048]
web_1     | django.channels.server INFO     HTTP GET /printers/4/?wizard=True 200 [0.04, 192.168.50.79:58048]
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/4/ 200 [0.04, 192.168.50.79:58048]
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/4/ 200 [0.04, 192.168.50.79:58048]
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/4/ 200 [0.04, 192.168.50.79:58048]
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/4/ 200 [0.04, 192.168.50.79:58048]
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/4/ 200 [0.04, 192.168.50.79:58048]
web_1     | django.channels.server INFO     HTTP GET /api/v1/octo/ping/ 200 [0.03, 192.168.50.168:49478]
web_1     | django.channels.server INFO     HTTP GET /api/v1/octo/ping/ 200 [0.02, 192.168.50.168:49482]
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/4/ 200 [0.04, 192.168.50.79:58048]
web_1     | django.channels.server INFO     HTTP GET /api/v1/octo/ping/ 200 [0.02, 192.168.50.168:49488]
web_1     | django.channels.server INFO     HTTP GET /api/v1/octo/ping/ 200 [0.02, 192.168.50.168:49490]
web_1     | django.request WARNING  Unauthorized: /api/v1/octo/ping/
web_1     | django.channels.server WARNING  HTTP GET /api/v1/octo/ping/ 401 [0.02, 192.168.50.168:49494]
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/4/ 200 [0.04, 192.168.50.79:58048]
web_1     | django.request WARNING  Unauthorized: /api/v1/octo/ping/
web_1     | django.channels.server WARNING  HTTP GET /api/v1/octo/ping/ 401 [0.02, 192.168.50.168:49498]
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/4/ 200 [0.04, 192.168.50.79:58048]
web_1     | django.channels.server INFO     HTTP GET /api/v1/octo/ping/ 200 [0.02, 192.168.50.168:49502]
[... 69 lines cut ...]
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/4/ 200 [0.04, 192.168.50.79:58048]
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/4/ 200 [0.04, 192.168.50.79:58048]
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/4/ 200 [0.04, 192.168.50.79:58048]
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/4/ 200 [0.04, 192.168.50.79:58048]
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/4/ 200 [0.04, 192.168.50.79:58048]
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/4/ 200 [0.04, 192.168.50.79:58048]
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/4/ 200 [0.04, 192.168.50.79:58048]
web_1     | django.channels.server INFO     HTTP POST /admin/app/printer/4/change/ 200 [0.12, 192.168.50.79:58187]
web_1     | django.channels.server INFO     HTTP GET /admin/jsi18n/ 200 [0.02, 192.168.50.79:58187]
web_1     | django.request WARNING  Not Found: /favicon.ico
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/4/ 200 [0.04, 192.168.50.79:58048]
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/4/ 200 [0.04, 192.168.50.79:58048]
[... cut ...]
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/4/ 200 [0.04, 192.168.50.79:58048]
web_1     | django.channels.server INFO     HTTP GET /admin/app/printer/ 200 [0.05, 192.168.50.79:58193]
web_1     | django.channels.server INFO     HTTP GET /admin/jsi18n/ 200 [0.02, 192.168.50.79:58193]
web_1     | django.request WARNING  Not Found: /favicon.ico
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/4/ 200 [0.05, 192.168.50.79:58048]
web_1     | django.channels.server INFO     HTTP GET /api/v1/printers/4/ 200 [0.04, 192.168.50.79:58048]
`

kennethjiang, 2020-12-23:
200 is the success code. it must be something else. Please turn on debug logging and check octoprint.log for errors: https://www.thespaghettidetective.com/docs/turn-on-debug-logging/

kennethjiang, 2021-01-11:
I can closing the issues that are actually supporting questions to discord server https://discord.gg/eE3QPVTk to streamlining the process.

closed 2021-01-11 by kennethjiang