TheSpaghettiDetective/obico-server #340: Odd scaling behavior, no full screen option

Kravchenko32, 2020-12-28:
https://imgur.com/a/cJYzA8D

kennethjiang, 2020-12-28:
This scaling behavior is the default of [Bootstrap](https://getbootstrap.com/), one of the most popular web framework to deal with different screen sizes.

Can you give us an example for a better scaling behavior?

Kravchenko32, 2020-12-29:
Sounds like bootstrap is hot garbage. I've already put in a lot of effort making that video for you and if you don't understand how it's an idiotic behavior from watching it I'm afraid I can't help you further.

kennethjiang, 2020-12-29:
Most users don't change the browser width when they are using the app. For them the default behavior is quite reasonable.

I appreciate your effort to make video for TSD. Unfortunately that won't be the reason for us to revamp our frontend framework just to cater to a use case that most of our users won't have.

closed 2020-12-29 by kennethjiang