TheSpaghettiDetective/obico-server #343: TSD listens on public interfaces

fischman, 2021-01-01:
The OctoPrint TSD plugin causes a janus process to open listening sockets on multiple ports on all interfaces:

```
$ sudo netstat -antlp|grep LISTEN | grep janus
tcp        0      0 0.0.0.0:8188            0.0.0.0:*               LISTEN      6600/janus
tcp6       0      0 :::8088                 :::*                    LISTEN      6600/janus
tcp6       0      0 :::7088                 :::*                    LISTEN      6600/janus
```

Naive attempts at hacking `OctoPrint/lib/python3.7/site-packages/octoprint_thespaghettidetective/bin/janus/etc/janus/janus.jcfg` (either `general.interface` or `nat.ice_enforce_list`) to restrict to either a particular external interface or the loopback interface both resulted in the edit getting overwritten automatically.

Is there a way to prevent this public listening?

kennethjiang, 2021-01-01:
This issue is probably for https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective.

But anyway, janus need to listen on the port to do the webcam streaming. If you don't want it, just disable premium webcam streaming in the settings and reboot the Pi and it should be gone. Let me know if it works and I'll close this issue.

fischman, 2021-01-01:
Disabling premium camera streaming and restarting did not help.

Considering from first principles it's unclear to me why a listener on public interfaces is required for webrtc-based camera streaming.
I'd expect all interfaces to get probed in order to construct the ICE candidate list, but listeners are not required for that.
Besides ICE candidate computation, janus exports an API which maybe the plugin uses over HTTP? But the plugin & janus running on the same machine means that janus would only need to listen on localhost/loopback interface for that to work (and, on linux at least, can get away even without that [using](https://janus.conf.meetecho.com/docs/FAQ.html#unixsockets) unix domain sockets). In fact seemingly the plugin already uses [`JANUS_SERVER = os.getenv('JANUS_SERVER', '127.0.0.1')`](https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/blob/master/octoprint_thespaghettidetective/webcam_stream.py#L37) so by default no public listening is needed?

LMK if you'd prefer I re-file this under the other project, or if what I say doesn't make sense.

kennethjiang, 2021-01-01:
By "public interface" do you mean any ip that is not "127.0.0.1"? If so, you've got a point. I assume it's janus default behavior to bind to all ip addresses as there is no point for us to make it so if otherwise.

If my understanding above about this issue is correct, I'll appreciate a PR to make janus bind to only localhost. Again this should be good enough as the plugin should always be running in the same Pi as janus.

fischman, 2021-01-02:
Indeed, not-127.0.0.1; this can be seen in the output of netstat above where the listening is happening on 0.0.0.0 and :: (the IPv4 and IPv6, resp., represenations of "all interfaces").

I am not previously familiar with janus, but peeking at the source code and playing around suggests that janus can be forced to use explicit IPs for these 3 ports by uncommenting & replacing the values with 127.0.0.1 or ::1 (for IPv4 or IPv6, resp.) at: [1](https://github.com/fischman/OctoPrint-TheSpaghettiDetective/blob/master/octoprint_thespaghettidetective/bin/janus/etc/janus/janus.transport.websockets.jcfg#L12), [2](https://github.com/fischman/OctoPrint-TheSpaghettiDetective/blob/master/octoprint_thespaghettidetective/bin/janus/etc/janus/janus.transport.http.jcfg#L21), and [3](https://github.com/fischman/OctoPrint-TheSpaghettiDetective/blob/master/octoprint_thespaghettidetective/bin/janus/etc/janus/janus.transport.http.jcfg#L46). Making those changes yields:
```
$ sudo netstat -antlp|grep LISTEN|grep janus
tcp        0      0 127.0.0.1:8088          0.0.0.0:*               LISTEN      5122/janus
tcp        0      0 127.0.0.1:8188          0.0.0.0:*               LISTEN      5122/janus
tcp6       0      0 ::1:7088                :::*                    LISTEN      5122/janus
```

(compare to netstat output in original report and note that the listening addresses are now explicitly the loopback ones).

Things I don't know:
- whether the janus.transport.http.jcfg:admin:admin_ip value should be hard-coded to IPv6 or not
- which of these 3 ports is actually used by the plugin (possibly one or more of these proposed edits is unnecessary and should be replaced instead by disabling the port outright)

kennethjiang, 2021-01-03:
Awesome! Mind sending a PR? Will merge once we got it tested.

closed 2021-01-04 by kennethjiang