TheSpaghettiDetective/obico-server #354: [BUG] Error when trying to run on an M1 Mac Mini

pdoronila, 2021-01-15:
**Describe the bug**
```
TheSpaghettiDetective % docker-compose up -d
Creating network "thespaghettidetective_default" with the default driver
Building ml_api
Step 1/7 : FROM thespaghettidetective/ml_api:base
base: Pulling from thespaghettidetective/ml_api
ERROR: Service 'ml_api' failed to build : no matching manifest for linux/arm64/v8 in the manifest list entries
TheSpaghettiDetective %
```

**To Reproduce**
Steps to reproduce the behavior:
1. cd TheSpaghettiDetective
2. docker-compose up -d

**Hosting environment (please complete the following information):**
 - OS: macOS Big Sir
 - Version 11.1
 - Hardware Mac Mini (M1, 2020)

[image]

kennethjiang, 2021-01-17:
We don't have a machine to build linux/arm64 image. Help will be appreciated!

pdoronila, 2021-04-10:
@kennethjiang im willing to help, what do I need to do?

kennethjiang, 2021-04-11:
You can help us by building the `ml_api` image for linux/arm64/v8 platform, and push it to our repo.

You can send email to me at k@thespaghettidetective.com so that I can give you push access for our docker hub.

closed 2022-04-26 by raymondh2