TheSpaghettiDetective/obico-server #361: [Feature] Option to move timelapse video storage to a custom directory

tonyzzz321, 2021-01-27:
**Is your feature request related to a problem? Please describe.**
Timelaspe videos are filling up my small server drive. I got a big harddrive but it seems like I cannot just move the timelapse video folder to the new drive without also moving the docker image folder.

**Describe the solution you'd like**
A way to specify the custom path for the tsd-timelaspe and tsd-pics directories would be nice.

**Describe alternatives you've considered**
To reclaim my precious main server drive storage:

I tried deleting the timelapse videos in the web interface, but it seems like the actual video files are not removed from my drive. Also I had to select and delete one by one. I wish there were a "select all" option. Ctrl-A didn't work for me.

I also tried deleting all files under the tsd-timelapses/private folder, but looks like it is not synced to the web interface, as the web interface still shows all the timelapse videos, but none of them can be played because the video files are gone.

kennethjiang, 2021-01-31:
This kind of features usually fall into "operations" category for a typical web backend like TSD. Usually it's a good practice is to not include operations toolchain in the web app itself to maintain the flexibility and to keep the web app clean. For instance, another user may want to archive the time-lapse to AWS S3. Yet another user may want to delete old time-lapse altogether.

I'd propose someone to start a separate repo for TSD private server Ops toolchain. But I'm open to other options.

charredchar, 2021-11-17:
I would agree with the option to select all and delete or some tool to check the folder for remaining videos and clean the interface of missing files. I periodically clean the files out as I don't need to store them long term. While a tool within TSD to do this would be nice I just a simple delete cronjob but they stay within the WebUI.