TheSpaghettiDetective/obico-server #364: [BUG] WebHook notifications provide incorrect URL

gstevenson, 2021-01-28:
**Describe the bug**
Discord (perhaps others) notifications provide a "Click Here to Examine" URL on the webhook notifications, but the URL appears to assume TSD is installed at the same IP address as the printer.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable Discord Webhook URL as per documentation
2. Send a print
3. Wait for it to complete
4. Click the "Click here to Examine" URL on the webhook notification

**Screenshots**

[image]

[image]

**Additional context**

I have TSD installed on a server at http://192.168.1.10:3334/printers/, and Octoprint at http://192.168.1.216/.

I'd expect to receive a notification giving me the first URL, but it gives me a URL of 192.168.1.216/printers

kennethjiang, 2021-01-29:
Hmmm I'm surprised. The server wouldn't know about your octoprint ip address so I'm wondering where that came from.

can you post the site configuration in your django admin console?

gstevenson, 2021-02-05:
Right you are! Configured wrong in the django console :( 

My apologies

closed 2021-02-05 by gstevenson