TheSpaghettiDetective/obico-server #366: [BUG] Home lan IP in "Please Confirm Your E-mail Address"

SA7BNT, 2021-02-01:
God morning. 
Just del all TSD docker img inklusive TSD git folder. 
After this i run "docker image prune" then git clone TSD and rebuild TSD docker. 
Config "docker-compose.yml" and build TSD docker. Config Django site to Domain, stopped the running TSD docker, rebuild with "sudo docker-compose up -d" But still getting my home Network IP in the "Please Confirm Your E-mail Address"
May this be a bug??

kennethjiang, 2021-02-01:
By "getting my home Network IP in the "Please Confirm Your E-mail Address"", do you mean in an email, or on the web page? A screenshot will be appreciated.

SA7BNT, 2021-02-01:
> By "getting my home Network IP in the "Please Confirm Your E-mail Address"", do you mean in an email, or on the web page? A screenshot will be appreciated.

Oh sorry absolut.
[image]

kennethjiang, 2021-02-01:
It should be your home IP address right? 

Do you mean this email shouldn't have been sent to begin with?

SA7BNT, 2021-02-01:
> It should be your home IP address right?
> 
> Do you mean this email shouldn't have been sent to begin with?

I'm wondering why if im config TSD with a domain i not getting the domain name in the mail instead of the home IP address. 

So something like 
https://my_domaine_name.com/accounts/confirm-email/MQ:1l6Tqv:hqVS1UM-OzL8B6AkM35VAE-EZQw/ 
instead of 
https://192.168.5.39:3334/accounts/confirm-email/MQ:1l6Tqv:hqVS1UM-OzL8B6AkM35VAE-EZQw/

kennethjiang, 2021-02-02:
The email is generated by `django-allauth`, which probably doesn't use django site in the way we expect it to.

Somebody needs to peel into `django-allauth` to figure this out.

SA7BNT, 2021-02-02:
Thanks for your replay

encetamasb, 2021-02-10:
https://github.com/pennersr/django-allauth/blob/32b42dfdbfb184f40f548f3ffccc58239e024a02/allauth/account/adapter.py#L440

@SA7BNT if you sign up at `https://yourdomain/` then you will see `yourdomain` in the confirmation email. Activation url is based on the actual url of the signup page instead of the configured django site.

closed 2021-02-10 by SA7BNT