TheSpaghettiDetective/obico-server #367: Webcam Streaming doesn't work on Mobile Network

mizz141, 2021-02-01:
**Describe the bug**
When I try to get Webcam Streaming working on my Phone with Mobile Data, the stream doesn't load nor give a black screen, I have already followed [this guide but it didn't help](https://www.thespaghettidetective.com/docs/webcam-feed-is-not-showing/)
However when I try with WiFi it works flawlessly, I'm connecting through Tailscale VPN.

**To Reproduce**
Steps to reproduce the behavior:
1. Start a Print
2. Connect to TSP with Phone through Chrome Browser and Tailscale on Mobile

**Screenshots**
[Screenshot from my Phone](https://imgur.com/a/Xl0GzmS)

**Hosting environment (please complete the following information):**
 - OS: unRAID
 - Version 6.9.0-rc2
 - Hardware PC

**Additional context**
I have set-up TSP with the help of [this guide ](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/docs/unraid_guide.md)

[Tailscale VPN](https://www.tailscale.com)

kennethjiang, 2021-02-01:
Can you try it with TSD cloud (https://app.thespaghettidetective.com/printers/) to see if it works? If it does, it means some routing/firewall rules for your private server stopped WebRTC to work probably.

mizz141, 2021-02-02:
When going over the TSP Cloud directly, it works properly.
But that also requires that I need premium for all other features that are behind a paywall, which is why I've set-up a private server in the first place.

idjmic, 2021-02-03:
I have the same issue and if I connect to my 3D Printer IP:CAM PORT I can see the camera, this is something to do with unRAID and blocking the camera image.

kennethjiang, 2021-02-04:
Do you see the 1-frame-per-10s basic streaming when you are printing? If so, it's just a WebRTC issue.

WebRTC has some intricate requirements for networking routing/firewall that renders it unusable in some cases. We have done a lot of extra work on routing in TSD cloud to make sure it's as compatible as possible. You can also do some trouble-shooting on your network to figure out why WebRTC doesn't work, but it'll require intimate knowledge on WebRTC.

closed 2022-04-26 by raymondh2