TheSpaghettiDetective/obico-server #368: [BUG] Error updating to latest version available in repository

SirSheik, 2021-02-02:
**Describe the bug**
A clear and concise description of what the bug is.
I am currently running TSD web version 1.3 (thespaghettidetective/web:base-1.3).
[image]

When I try to update and compile the new version to work with the latest version of the Octoprint plugin, I get the following error. 

```
Step 7/7 : RUN wget --quiet -O model/model.weights $(cat model/model.weights.url | tr -d '\r')
 ---> Running in df490e3e3754
ERROR: Service 'ml_api' failed to build : The command '/bin/sh -c wget --quiet -O model/model.weights $(cat model/model.weights.url | tr -d '\r')' returned a non-zero code: 4
```

This error is displayed when downloading new files to an existing folder and to a new folder.

**To Reproduce**
First scenario on an existing folder:
git pull
sudo docker-compose up --build -d
[First scenario output.txt](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/files/5912760/First.scenario.output.txt)

Second scenario on a new, empty folder:
git clone https://github.com/TheSpaghettiDetective/TheSpaghettiDetective.git
cd TheSpaghettiDetective && sudo docker-compose up -d
[Second scenario output.txt](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/files/5912759/Second.scenario.output.txt)

**Screenshots**
[image]

**Hosting environment (please complete the following information):**
 - OS: Linux  
 - Version: Ubuntu 20.04.1 LTS GNU/Linux 5.4.0-65-generic x86_64)
 - Hardware: Cloud server (VPS)

**Additional context**
On previous occasions I was able to update the version following the same steps without errors.

Thank you very much

kennethjiang, 2021-02-05:
Thank you for the detailed report and sorry for the delay.

To further debug, please copy this hash code from your console:

[image]

Then run these commands using the hash code you grabbed:

[image]

Let me know what you see in the last command.

SirSheik, 2021-02-05:
Hi @kennethjiang 
Thank you very much for your answer and your help.
I executed the command inside the container but after waiting I did not get back any result.
`wget --quiet -O model/model.weights $(cat model/model.weights.url | tr -d '\r')`

This is the output:
```
root@MyCloudServer:~/TheSpaghettiDetective# sudo docker-compose up --build -d
WARNING: The DOCKERFILE_VERSION variable is not set. Defaulting to a blank string.
Building ml_api
Step 1/7 : FROM thespaghettidetective/ml_api:base
 ---> 6eaa0374427c
Step 2/7 : WORKDIR /app
 ---> Using cache
 ---> 857f1af0daac
Step 3/7 : EXPOSE 3333
 ---> Using cache
 ---> 774c04bb4a72
Step 4/7 : ADD . /app
 ---> Using cache
 ---> e458421f5eae
Step 5/7 : RUN pip install --upgrade pip
 ---> Using cache
 ---> 1baffb9402f2
Step 6/7 : RUN pip install -r requirements.txt
 ---> Using cache
 ---> c52adc06d8c7
Step 7/7 : RUN wget --quiet -O model/model.weights $(cat model/model.weights.url | tr -d '\r')
 ---> Running in c29f3b65132f
ERROR: Service 'ml_api' failed to build : The command '/bin/sh -c wget --quiet -O model/model.weights $(cat model/model.weights.url | tr -d '\r')' returned a non-zero code: 4
root@MyCloudServer:~/TheSpaghettiDetective# docker run --rm -it c52adc06d8c7 bash
root@6e327195b2af:/app# ls
Dockerfile  Dockerfile.aarch64  Dockerfile.base  auth.py  bin  lib  model  requirements.txt  requirements_x86_64.txt  scripts  server.py  wsgi.py
root@6e327195b2af:/app# wget --quiet -O model/model.weights $(cat model/model.weights.url | tr -d '\r')
```

kennethjiang, 2021-02-05:
What did you see after this command?

[image]

SirSheik, 2021-02-05:
Hi @kennethjiang 
I'm sorry but the command you say doesn't return any results. It takes a couple of seconds to process but there is no result.
You can see it better with the next screenshot.
[image]

kennethjiang, 2021-02-05:
Quite strange. it looks like the wget command actually succeeded.

Please run these commands so that I can dig in more.

[image]

SirSheik, 2021-02-05:
It returns the value 4

```
root@a0bf02f87c21:/app# wget --quiet -O model/model.weights $(cat model/model.weights.url | tr -d '\r') echo $?
root@a0bf02f87c21:/app# echo $?
4
```
As the value that displays the error during build progress.
[image]

kennethjiang, 2021-02-07:
Sorry have been out in the past 2 days.

It looks like the wget command was not successful. please run `wget -O model/model.weights $(cat model/model.weights.url | tr -d '\r')` instead.

SirSheik, 2021-02-07:
Thank you very much, thanks to your last test I detected that there was some connection problem inside the container. After researching and testing, the problem was that my firewall rules did not include the "bridge" network which seems to be temporarily used by the container during the build process.
After solving the network problem everything compiled fine.
Thank you again for your time and help.

closed 2021-02-07 by SirSheik