TheSpaghettiDetective/obico-server #369: Use App with local server

tbensonwest, 2021-02-04:
Hi

Apologies, but I couldn't find an answer to this elsewhere.

Is it currently possible to point the Android app at a local TSD server? Or, is the app source code available to build with a custom server IP?

Great work btw. Really appreciate what you have built!

kennethjiang, 2021-02-04:
It is in the works. It may not come very soon as we are working through other priorities of the app. Will update you guys in the monthly newsletter. Stay tuned.

closed 2021-02-04 by kennethjiang

thehoff, 2021-03-07:
Is the app code open source too? could this be a community input as this is important to me.

kennethjiang, 2021-03-08:
Opening the source code for the app will take much longer, as it's much trickier to abstract the credentials, such as keys, ids, certificates, etc from mobile app than from a web app (basically just putting credentials in env vars).

Help will be appreciated if someone is familiar with how this can be done.

thehoff, 2021-03-08:
I'm not really an app dev other than pulling apps apart for code testing. I could probably work it out but I'm so time poor I couldn't commit to anything :(

 It was just a thought as the self hosted is amazing and I'm just sitting on my VPN and the webUI which works fine.

Was just curious.