TheSpaghettiDetective/obico-server #370: [BUG] Access to mjpg-streamer page fails

charredchar, 2021-02-04:
**Describe the bug**
mjpg-streamer webpage unable to load with TSD OctoPrint Plugin enabled. Attempting to load the control.htm page to adjust webcam, unable to access while TSD Plugin is enabled. Disabling TSD Plugin allows access to the page once again. This page is required for other plugins.

**To Reproduce**
Steps to reproduce the behavior:
1. Install TSD Plugin within OctoPrint.
2. Enabled mjpg-streamer page access in OctoPi.
3. Attempt to load mjpg-streamer webpage via ip/webcam or ip:8080.

**Hosting environment (please complete the following information):**
 - OS: OctoPi 0.17
 - Version: 1.5.0
 - Hardware: Raspberry Pi 4 1GB

**Additional context**
Attempting to access the main page for mjpg-streamer returns error 400. Attempting to load the control.htm page returns a 404.

kennethjiang, 2021-02-05:
This was a design choice that we had to make in order to do implement premium webcam streaming, since mjpeg-streamer is using mjpeg encoding, which is incompatible with the H.264 encoding that TSD premium streaming uses.

You can disable premium streaming if mjpeg-streamer page is important for you: https://www.thespaghettidetective.com/docs/disable-25-fps-streaming/

Unfortunately we can't do anything to keep premium streaming and mjpeg-streamer page on at the same time. So I'll close this issue. But you are welcome to add more comments if you believe there are options that we haven't explored before.

charredchar, 2021-02-05:
This is amazing information, thank you so much for going into detail to explain why it is the case and how to resolve it! I am happy to see there is a method to choose to bring it back as in my case I don't stream from TSD interface anyway so I do not need premium steaming.

I wouldn't know of a method to keep both, but I've been through the settings and documents a few times in the last few versions and never noticed a comment on loosing access to the mjpg-streamer page if premium is enabled. Maybe add this to Reason #2 on the page you linked, I know it is listed as an issue in OctoLapse that TSD breaks Camera Adjustments and this is a resolution to that issue.

kennethjiang, 2021-02-05:
We assume most OctoPrint users don't directly access mjpg-streamer page and that's why we didn't document the fact the premium streaming will break it.

Just out of curiosity, what is you use case to access the mjpg-streamer page, instead of the mjpeg stream itself (we do maintain compatibility with the mjpeg stream)?

charredchar, 2021-02-13:
> We assume most OctoPrint users don't directly access mjpg-streamer page and that's why we didn't document the fact the premium streaming will break it.

I would agree it is a safe assumption that most wouldn't but can become a headache for anyone trying to resolve an issue.

> Just out of curiosity, what is you use case to access the mjpg-streamer page, instead of the mjpeg stream itself (we do maintain compatibility with the mjpeg stream)?

In the first case it was OctoLapse that needs direct access to the page to modify and apply camera settings via the user interface. If the page is not accessible it breaks the ability to do so. You can still access the stream but won't be able to modify settings.

https://github.com/FormerLurker/Octolapse/wiki/V0.4---Enabling-Camera-Controls

kennethjiang, 2021-02-14:
I didn't know OctoLapse actually directly reaches to mjpeg-streamer page to change its settings. But it's a legitimate use case. I have changed the title to reflect this, and opened this issue. However, as overwhelmed as we are right now, this issue may not be picked up soon. Sorry about that.

Doormat1, 2021-03-10:
Hi, Just reading this, so I understand you cant use premium streaming with octolapse? This would explain the camera feed drops the moment I start to print! Ive disabled octolapse and the camera feed seems to behave it self.