TheSpaghettiDetective/obico-server #376: send Email not working on Gmail

matrix1995, 2021-02-11:
Hey,
i´m using TSD on my own Windows Server 2019 but i can´t send emails.
I activated SMTP in Gmail and i added my email in the docker-compose file.

Is there somethig else i have to do?

docker-compose:

.............................................................
    EMAIL_HOST:'smtp.gmail.com'
    EMAIL_HOST_USER:'mymail@gmail.com'
    EMAIL_HOST_PASSWORD:'mymailpassword'
    EMAIL_PORT: 587  # DO NOT surround it with quotes. Otherwise email won't be sent!
    EMAIL_USE_TLS: 'True'
    DEFAULT_FROM_EMAIL:'mymail@gmail.com'
    DEBUG: 'False'    # Don't set DEBUG to True, otherwise the static files will be cached in browser until hard-refresh
    SITE_USES_HTTPS: 'True' # set it to 'True' if https is set up
    SOCIAL_LOGIN: 'False'
    REDIS_URL: 'redis://redis:6379'
    DATABASE_URL: 'sqlite:////app/db.sqlite3'
    INTERNAL_MEDIA_HOST: 'http://web:3334'
    ML_API_HOST: 'http://ml_api:3333'
    ACCOUNT_ALLOW_SIGN_UP: 'False'  # -> set to 'True' if you want to open sign up form
    WEBPACK_LOADER_ENABLED:
.......................................................

Jlrine2, 2021-02-12:
I'm also having issues with email sending, I'm getting this in the logs
```
WARNING  Email settings are missing. Ignored send requests
```
Are you by chance also getting this message, you can check the logs with `docker-compose logs`

kennethjiang, 2021-02-12:
This is likely a configuration issue. Have you guys asked for help on our discord server: https://discord.gg/zvSzJea6as

matrix1995, 2021-02-12:
I get the same error as you @Jlrine2.

I asked in the Discord for help.
If i get a solution I will write it here.

s4b3rt0oth, 2021-03-23:
I was finally able to get email working. I went into django admin and gave my user all of the permissions. (I'm sure only the email permissions were needed but I did them all anyway. I'll do more testing later.) 

I'm not sure having "superuser" checked did it by itself, but with a similar config as the one posted this is what worked for me. 

Also, just in case, make sure that in Gmail you setup an app password and don't use your actual account. I think Gmail blocks extraneous logins with SMTP.

closed 2022-04-26 by raymondh2