TheSpaghettiDetective/obico-server #381: [Feature] Add the ability to use TSD with Duet 3D printer boards

tobiasvogel, 2021-02-16:
The TSD has been developed as a OctoPrint-plugin.
It would be great to be able to use TSD with Duet printer boards. 

A possible approach would be having TSD run in some kind of standalone mode
where it would be fed with a camera-stream (e.g. by supplying a stream-url) and
would simply call some user-defined function to pause a print (like calling some url).

DuetWebControl (DWC) integrates webcam-streams like this anyway and the
DuetWebServer provides a RESTful API that could be called to interface with the
board.

raymondh2, 2021-02-17:
This will still need another computer to run the webcam to TSD server part of the application. Defeating most of the point of not using Octoprint.

ecksate, 2021-03-30:
I'd also like to use the spaghetti detective, and use duet's software.  In any way, just to get email notifications or something

ecksate, 2021-03-30:
Is there a way to put webcam video into a directory and run TSD on it?

raymondh2, 2021-03-30:
You can upload a time lapse to TSD to process it. The best current solution it to run TSD on Octoprint. As no matter what using TSD with a duet will require a Linux computer of some sort. And running TSD on the duet directly is not possible primary due to the fact we would need to maintain a fork of Reprapfirmware and second because of the low power MCUs used.

tobiasvogel, 2021-03-30:
I am currently working on a rather dirty "hack" of sorts to achieve this: I have the TSD-Docker instance running on a regular PC and am using a Raspberry Pi in combination with my Duet. My goal - at least as a first step - is to create kind of a stand-in replacement which takes in the webcam stream from the Duet Web-Interface and feeds it into Octoprint from one side and interacts from there (Octoprint) with the Duet using the Duets API by calling G-Code macros on the opposite side.
In a further step I plan to eliminate Octoprint from the equation and write a bridge to directly pass-through the data from the Duet to TSD and vice-versa.
I know that this will by no means provide a proper implementation, however I hope to achieve a somewhat usable way of interfacing the two. I'll share more insight and/or updates here as things progress.

tobiasvogel, 2021-03-30:
Sure, that is definitely the case - however, at least it would introduce one option at all for Duet users like myself.. :)

psychotik2k3, 2023-01-17:
That plugin would be possible only in SBC mode to use the pi usage, and so we could have the camera stream on the pi too. the tricky part is to have an octoprint plugin wrapper