TheSpaghettiDetective/obico-server #389: [BUG] Docker still ignoring Override .yml for Jetson Nano 2GB

Ethycs, 2021-02-22:
**Describe the bug**
Documentation between the github and https://www.thespaghettidetective.com/blog/2020/04/19/jetson-nano-spaghetti-detective-guide/ still does not match up.

**To Reproduce**
Steps to reproduce the behavior:
1. Examine the GitHub Docker Readme
2. Attempt to follow the instructions on the Github jetson link

 - OS:  Ubuntu 18.04.5 LTS
 - Version [e.g. 22]
 - Hardware Jetson Nano 2GB
 
Please clarify this issue for GitHub side documentation.

kennethjiang, 2021-02-23:
Sorry for the delay!

I don't have a Nano (probably it's the time to get one!). So I'll leave @LyricPants66133 or @raymondh2 to comment on this issue.

raymondh2, 2021-02-23:
I'll look into this but the 2gb nano is not supported due to the ram being shared with the gpu

Ethycs, 2021-02-26:
Would be great to have that listed somewhere, I grabbed a Jetson for the sole purpose of TSD.

raymondh2, 2021-02-26:
https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/docs/jetson_guide.md

Ethycs, 2021-02-26:
> https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/docs/jetson_guide.md

Productive contribution

https://github.com/TheSpaghettiDetective/TheSpaghettiDetective#hardware-minimum-spec
"If you have an old PC with at least 2GB of memory, you will be probably be fine."

kennethjiang, 2021-02-27:
@Ethycs sorry for the confusion. There reason why jetson needs more memory than a PC is because it shares video RAM with main RAM.

But in any case, I just changed the README to make the min requirement to 4GB. Don't think many people still have a 2GB PCs.

Not sure why `docker-compose.override.yml` seems to work for some Jetson people but not others...

closed 2022-04-26 by raymondh2