TheSpaghettiDetective/obico-server #392: [Feature] Notifications when printer sends action commands

Doormat1, 2021-02-26:
I have recently installed the smart filament sensor on my ender 3 pro runnin an skr mini e3 v2. 
I have host commands enabled in my firmware so when the printer sends the m600 command when the sensor detects an issue octoprint pauses. 
Is there anyway TSD could notify me when these happen as well?

kennethjiang, 2021-02-27:
This is a really good feature request.

Does OctoPrint pick up these m600 command, and hopefully even make it available with its APIs? If so, it'll be a lot easier to implement. Will appreciate it if you can post a few more pointers about how widely accepted m600 command is, and the status of OctoPrint support of that.

Doormat1, 2021-02-27:
Yes Octoprint picks them up and pauses the print with a notification on the
screen until the user deals with it.

kennethjiang, 2021-02-27:
That's a good news!

Can you dig up a bit more info, such as if OctoPrint will send an event to all plugins on m600? I don't have a printer that I can test with right now.

Doormat1, 2021-02-27:
This is the octoprint documentation
https://docs.octoprint.org/en/master/features/action_commands.html
https://docs.octoprint.org/en/master/bundledplugins/action_command_prompt.html

hope they help

kennethjiang, 2021-02-27:
This is awesome! Should make things a lot easier! Thank you @Doormat1

Doormat1, 2021-02-27:
Glad to help! People need to enable it in config adv, I got the
instructions from the teaching tech video on the smart filament sensor

kennethjiang, 2021-02-27:
Gotcha. Thanks again for the tips so that I don't have to waste time searching for info. :)

dfredell, 2021-09-19:
My use of m600 is slightly different. I sometimes do a two colored print on my Ender 3 and switch the filliment part way through the print. The M600 is added to the gcode via Cura at layer X. 
Currently I have Octoslack installed to notify me of M600s.

kennethjiang, 2021-09-20:
This is a really good info @dfredell . Will definitely give us a jump start!