TheSpaghettiDetective/obico-server #401: [Feature] Is it possible to access the real-time data of the current state of the printing through api?

Deymos42, 2021-03-04:
I would like to incorporate the error value of the current print into an octoprint view that I am developing. But from what I have seen to access this data on the TSD website it is done through socket (ws://localhost:3334/ws/web/1/).

Is there an easy way to access this data?

Thanks in advance

kennethjiang, 2021-03-05:
I just tried `http://localhost:3334/api/v1/printers/{printer_id}/` and it returned the same info as it would in in the websocket. Please try it to see if it works for you.

kennethjiang, 2021-03-05:
Obviously you need to pass the correct cookie to authenticate the api call

Deymos42, 2021-03-05:
Ou, its true. I can use normalized_p, I had not seen it. Thank you very much, I close.

closed 2021-03-05 by Deymos42