TheSpaghettiDetective/obico-server #403: The ability to manage uploaded files in folders

brynnflynn, 2021-03-05:
**Is your feature request related to a problem? Please describe.**
I have two different kinds of printers, soon to be a third (3 ender 5's, a CR10s, and about to add an Ender 5 Plus). Obviously, sliced files have different settings between all of these, and keeping track of a big list of files is a bit of a pain to say the least. 

**Describe the solution you'd like**
I'd like to be able to create folders for and organize uploaded GCODE into them through the app and the website. Ideally I should be able to navigate through the same folders between the two. Ideally the same structure would be followed on the Octoprint instance I'm controlling, but this is less important to me than making sure I don't send GCODE meant for a CR10 to an ender 5.

**Describe alternatives you've considered**
This functionality doesn't exist so there's not many options. I already use a naming convention that calls out which machine in the prefix to the file name, but since file names don't show entirely on the app and have a ..., it makes it more difficult to identify which file I'm printing.

**Additional context**
Picture from my file list; you can see I prefix things with the machine it's intended for, but if I'm rushing I can skip over it by accident. Having folders would add a step to starting a print, but would help me track and organize what I'm doing better!
[image]
.

brynnflynn, 2021-03-10:
And regarding the file name prefix, my file names as you saw are fairly long, and when I'm trying to get a print started quickly it's easy to gloss over a list with "CE5_...., CE5_...., CE5P_..., CE5_..." and accidentally select the CE5P. If I only have 3 folders to pick from, it's much easier to avoid that problem and get into the right context so to speak.

kennethjiang, 2021-03-10:
Thank you so much for the additional feedback! Definitely helpful in providing the info for us to come up with a design that's intuitive for most users!

brynnflynn, 2021-03-19:
[image]
What happens when you can't filter files by the printer type and helpful assistant starts a CR10S print on an Ender 5. XD