TheSpaghettiDetective/obico-server #404: [Feature] Use of github tags/releases to identify stable versions

sylvaneau, 2021-03-06:
Hi,

I'm using TSD deployed on a private server. First of all, congrats ! It works very well.
I'm following your work and updating my server quite often.

It would be nice to have some tags/releases on this repo (as you already do in the octoprint plugin repo).
It could be a nice way to identify stable versions of the server and to synchronise it with matching versions on the octoprint plugin.

Once again thx for the amazing job !

kennethjiang, 2021-03-06:
Thank you for making this request.

We have versioning for our plugin. However, for the server side, we have been following the CI/CD practice, which enables us to move a lot faster than otherwise.

The gist of CI/CD is to keep `master` branch always releasable. I thought we have done a fair job at that (maybe I'm over confident here lol). We occasionally broke the master branch but we almost always fixed it within 24 hours, which is the longest release cycle that we have.

The server can't lock-step with the plugin versioning, as the server needs to make sure older (actually much older, as some users are still running 0.x plugin) versions of the plugin still work. 

Has the lack of server versioning caused any problem for you? If so, can you elaborate on what the problem is?

sylvaneau, 2021-04-12:
Hi, very sorry for answering so late,

You're actually totally right, I didn't figure out that you were following a GitFlow like workflow. And you're right, I never had any issue with an update mismatch.

You can definitely close this issue, sorry for making this "noise".

Ahaha this issue number is #404, kind of a sign :-D

closed 2021-04-12 by sylvaneau

kennethjiang, 2021-04-13:
No problem. Thank you for following up on this issue!