TheSpaghettiDetective/obico-server #406: [BUG] docker-compose fails to bring up website

paxxoid, 2021-03-09:
I previously installed via docker without issue and today when I went to view the website was met with strange behavior. a view of the logs displayed that the database was locked:

```
django.request ERROR    Internal Server Error: /api/v1/octo/pic/
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
  File "/usr/local/lib/python3.7/site-packages/django/db/backends/sqlite3/base.py", line 383, in execute
    return Database.Cursor.execute(self, query, params)
sqlite3.OperationalError: database is locked
```

I brought down the containers, rebuilt and tried again but the database was never released no matter what I did. As a last resort, I removed all containers/images/volumes for the TSD, cloned a fresh copy, and attempted to bring up. I am now being met with the error and an unresponsive web interface.

```

itsme@home:/mnt/raid/TheSpaghettiDetective$ docker logs -f 9cdbc73f4dc9

0 static files copied to '/app/static_build', 278 unmodified, 564 post-processed.
/usr/local/lib/python3.7/site-packages/daphne/server.py:15: UserWarning: Something has already installed a non-asyncio Twisted reactor. Attempting to uninstall it; you can fix this warning by importing daphne.server early in your codebase or finding the package that imports Twisted and importing it later on.
  UserWarning,
```

During the build process

```

Deleting 'css/app.css.gz'
Deleting 'css/app.css'

278 static files copied to '/app/static_build', 798 post-processed.
/usr/local/lib/python3.7/site-packages/daphne/server.py:15: UserWarning: Something has already installed a non-asyncio Twisted reactor. Attempting to uninstall it; you can fix this warning by importing daphne.server early in your codebase or finding the package that imports Twisted and importing it later on.
  UserWarning,
Removing intermediate container a4ec5b2078c2
 ---> a9d3dac5da42

Successfully built a9d3dac5da42
Successfully tagged thespaghettidetective_web:latest

```
any ideas?

kennethjiang, 2021-03-09:
@paxxoid This looks like an environment issue. You will get help much faster by asking this question in our discord server: https://discord.gg/NcZkQfj

Let me know if you really think this is a code issue. Otherwise I'll close it assuming it's an environment/configuration question.

paxxoid, 2021-03-10:
Going to close this issue  for now- I was able to launch the docker image yesterday night and while I was receiving database lock errors initially, I have not seen any today when I monitored. 
Thanks

closed 2021-03-10 by paxxoid