TheSpaghettiDetective/obico-server #420: Exec format error (Same as #187)

doggogoesbork, 2021-03-26:
Brand new Jetpack45 on 2gb Jetson nano, 32 gb sd card. Only setting modified was switching the ip address to static

running this command

git clone https://github.com/LyricPants66133/Jetson_TSD_Fullinstall.git && sh Jetson_TSD_Fullinstall/jetson_TSD_install.sh

Eventually leads to the exact same error as #187 

`Step 4/7 : ADD . /app

---> 85f6d8619a58

Step 5/7 : RUN pip install -U pip

---> Running in 93d948304785

standard_init_linux.go:211: exec user process caused "exec format error"

ERROR: Service 'tasks' failed to build: The command '/bin/sh -c pip install -U pip' returned a non-
zero code: 1``

kennethjiang, 2021-03-27:
if I remember correctly @raymondh2 said 2GB nano won't work.

therealj4nd3rs0n, 2021-03-27:
I am having the exact same issue, except I have the 4GB jetson nano, 32g SD card.  I modified absolutely nothing and followed the exact instructions found here:  https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/docs/jetson_guide.md

kennethjiang, 2021-03-27:
@LyricPants66133 Any chance you know why this is the case?

therealj4nd3rs0n, 2021-03-27:
Well, it isn't a problem in the requirements.txt file, I commented out everything in there and even removed it, same error.  Just in case, I also deleted the x86 shared libs from the ml_api/bin directory since other google searches on this specific error mentioned architecture incompatibility as a source for that error.  I am nobody's idea of a docker-compose expert, but I haven't given up yet.

LyricPants66133, 2021-03-30:
Thanks for the ping Kenneth. I think I may have a solution. Try downgrading your Jetpack SDK version to [4.3](https://developer.nvidia.com/jetpack-43-archive), and try the installation process again. Please tell me how it goes.

therealj4nd3rs0n, 2021-03-30:
@LyricPants66133 I tried your suggestion, unfortunately with the exact same result.

LyricPants66133, 2021-03-30:
@therealj4nd3rs0n I found a small mistake in the install script! I just fixed it. I tried running the install on my jetson and it seemed to work, but this could just be my correctly configured Ubuntu. Can you try again?

therealj4nd3rs0n, 2021-03-31:
@LyricPants66133 I have bad news and good news.  First the (small) bad news.  Your updated override_file_writer.py script contains a small error.  On line 9, you have an extra period at the end of 'Dockerfile'.  Instead of 'Dockerfile.' it should just be 'Dockerfile' 
As it is, the script doesn't edit the Dockerfile to put in the arm version, instead it creates a new file named 'Dockerfile.' in that directory, see this 'ls' output:
api  app  config  Dockerfile  Dockerfile.  Dockerfile.base  Dockerfile.dev  frontend  lib  manage.py  requirements.txt

Now for the very good news.  I manually edited the 'Dockerfile' with what your script was supposed to do, manually re-ran 'sudo docker-compose up -d' and it works!  Sorry, I should have figured out the issue on my own, I was on the right track and just did not quite find it due to my unfamiliarity with docker-compose.  Thank you for all of your help, sincerely appreciated!!

LyricPants66133, 2021-03-31:
@therealj4nd3rs0n oh well that's embarrassing. That '.' Is left over from an unnecessary file ending that was there before..... Well, I'm glad that you were able to get everything working this time.

kennethjiang, 2021-03-31:
Thank you so much @LyricPants66133 for your help here.

Was this issue fixed by making changes in your repo, or some of the files in the TSD repo needs to be changed as well?

therealj4nd3rs0n, 2021-03-31:
@kennethjiang I will reply quick since I saw your question.  No changes to anything in the TSD repo.  @LyricPants66133 has updated his script so that the docker-compose step will pull the correct container images down.  You should now be able to copy/paste the single line from the Jetson Nano readme file referenced at the top of this thread and get a successful result.

kennethjiang, 2021-03-31:
I will close this issue then. Glad you guys figured this out!

closed 2021-03-31 by kennethjiang