TheSpaghettiDetective/obico-server #424: [BUG] (2 gb Jetson Nano P45)  ModuleNotFoundError: No module named 'corsheaders'

doggogoesbork, 2021-04-02:
Fresh install, used the following script:
https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/docs/jetson_guide.md

after the recent fixes, I can load the dockers and they are running, however no web page ever comes up. Opening the logs of the web container reveals this error, repeating constantly

ModuleNotFoundError: No module named 'corsheaders'
/usr/local/lib/python3.7/site-packages/daphne/server.py:15: UserWarning: Something has already installed a non-asyncio Twisted reactor. Attempting to uninstall it; you can fix this warning by importing daphne.server early in your codebase or finding the package that imports Twisted and importing it later on.
  UserWarning,
Traceback (most recent call last):
  File "manage.py", line 15, in <module>
    execute_from_command_line(sys.argv)
  File "/usr/local/lib/python3.7/site-packages/django/core/management/__init__.py", line 381, in execute_from_command_line
    utility.execute()
  File "/usr/local/lib/python3.7/site-packages/django/core/management/__init__.py", line 357, in execute
    django.setup()
  File "/usr/local/lib/python3.7/site-packages/django/__init__.py", line 24, in setup
    apps.populate(settings.INSTALLED_APPS)
  File "/usr/local/lib/python3.7/site-packages/django/apps/registry.py", line 91, in populate
    app_config = AppConfig.create(entry)
  File "/usr/local/lib/python3.7/site-packages/django/apps/config.py", line 90, in create
    module = import_module(entry)
  File "/usr/local/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'corsheaders'

kennethjiang, 2021-04-03:
This doesn't look like a code issue. Can you go to our private server channel in https://discord.gg/NcZkQfj and post this question? You will get faster response. Plus people there can ask you run a debugging commands to help you more effectively.

sashcardi, 2021-04-20:
@doggogoesbork did you ever resolve this? It's not code related but I have come across the exact same issue on my Jetson Nano. Just wondered if you ever got it sorted.

hendrikreuhl, 2021-05-20:
Hi,

i am actually having the same issue. Did you found a proper solution for this issue?

Thanks

RobeeeJay, 2021-06-05:
Has anyone managed to fix this issue? I just ran into it today 😞

LyricPants66133, 2021-06-07:
@RobeeeJay @hendrikreuhl @doggogoesbork @sashcardi This issue has been fixed. I am now @kennethjiang to resolve this issue.

closed 2021-06-07 by kennethjiang