TheSpaghettiDetective/obico-server #426: SMTP missing

drveni, 2021-04-05:
hi

i got vm machine with ubuntu server 20.04 . on server i have docker, and containers, and works ok.

i make new container with instruction from github. ok spaghetti server works. problem is mail server for sending smtp. python3 missing "smtplib" ..on server i really dont have in that directoy .. im try install whit pip and pip3... and still wont work/install correct smtplib. any correct commad to install python smtplib module? i look on python webpage but no luck

"python3 -m pip install smtplib"

i know its module but how to install correct and work whit spaggeti

ERROR: Could not find a version that satisfies the requirement smtplib (from versions: none)
ERROR: No matching distribution found for smtplib
tiho@ubuntuserver:/usr/local/lib/python3.8$ 
"
could you import this module to next update !?  or how to manual install?

this is log file from portainer

2021-04-05T14:06:13.805116168Z   File "/usr/local/lib/python3.7/smtplib.py", line 359, in send
2021-04-05T14:06:13.805121977Z     raise SMTPServerDisconnected('please run connect() first')
2021-04-05T14:06:13.805125460Z smtplib.SMTPServerDisconnected: please run connect() first
2021-04-05T14:06:13.812194461Z django.channels.server ERROR    HTTP POST /accounts/email/ 500 [0.02, 192.168.5.68:60303]
2021-04-05T14:06:18.736017921Z django.channels.server INFO     HTTP POST /accounts/email/ 302 [0.04, 192.168.5.68:60303]
2021-04-05T14:06:18.760209147Z django.channels.server INFO     HTTP GET /accounts/email/ 200 [0.02, 192.168.5.68:60303]
2021-04-05T14:06:23.333749094Z django.request ERROR    Internal Server Error: /accounts/email/
2021-04-05T14:06:23.333767084Z Traceback (most recent call last):
2021-04-05T14:06:23.333771153Z   File "/usr/local/lib/python3.7/site-packages/django/core/handlers/exception.py", line 34, in inner
2021-04-05T14:06:23.333774723Z     response = get_response(request)

kennethjiang, 2021-04-06:
This is what we have for TSD cloud server:

```
bash-5.0# pip freeze
aioredis==1.3.1
amqp==2.6.1
asgiref==3.2.10
async-timeout==3.0.1
attrs==20.2.0
authy==2.2.3
autobahn==20.7.1
Automat==20.2.0
backcall==0.2.0
backoff==1.10.0
billiard==3.6.3.0
bson==0.5.10
CacheControl==0.12.6
cachetools==4.1.1
celery==4.4.7
certifi==2020.6.20
cffi==1.14.3
channels==2.4.0
channels-redis==3.0.1
chardet==3.0.4
chargebee==2.8.2
constantly==15.1.0
cryptography==3.1.1
daphne==2.5.0
decorator==4.4.2
defusedxml==0.6.0
discord-webhook==0.8.0
dj-database-url==0.5.0
Django==2.2.10
-e git://github.com/kennethjiang/django-admin-charts.git@ad097ae82e14d7836be2d4fe97a70b7f40091838#egg=django_admin_charts
django-admin-rangefilter==0.6.1
django-allauth==0.44.0
django-bootstrap-pagination==1.7.0
django-bower==5.2.0
django-channels-presence @ git+https://github.com/ivahotin/django-channels-presence.git@8cc0775adcd653c4bd920294c4e6f3d980692d40
django-compat==1.0.15
django-cors-headers==3.6.0
django-debug-toolbar==2.0
django-extensions==2.1.4
django-jsonfield==1.4.1
django-jstemplate==1.3.8
django-nvd3==0.9.7
django-qr-code==1.2.0
django-qsstats-magic==1.1.0
django-safedelete==0.5.1
django-settings-export==1.2.1
django-simple-history==2.7.0
django-webpack-loader==0.7.0
django-widget-tweaks==1.4.3
djangorestframework==3.9.1
djcacheutils==3.0.0
firebase-admin==4.4.0
google-api-core==1.22.4
google-api-python-client==1.12.3
google-auth==1.22.1
google-auth-httplib2==0.0.4
google-cloud-core==1.4.3
google-cloud-firestore==1.9.0
google-cloud-storage==1.31.2
google-crc32c==1.0.0
google-resumable-media==1.1.0
googleapis-common-protos==1.52.0
grpcio==1.32.0
gunicorn==19.9.0
hiredis==1.1.0
httplib2==0.18.1
hyperlink==20.0.1
idna==2.10
importlib-metadata==2.0.0
incremental==17.5.0
ipdb==0.13.4
ipython==7.18.1
ipython-genutils==0.2.0
jedi==0.17.2
Jinja2==2.11.3
jsonfield==2.0.2
kombu==4.6.11
MarkupSafe==1.1.1
msgpack==1.0.0
newrelic==4.12.0.113
oauth2client==4.1.3
oauthlib==3.1.0
parso==0.7.1
pexpect==4.8.0
phonenumbers==8.10.5
pickleshare==0.7.5
Pillow==5.4.1
prompt-toolkit==3.0.7
protobuf==3.13.0
psycopg2==2.7.7
ptyprocess==0.6.0
pushbullet.py==0.11.0
pyasn1==0.4.8
pyasn1-modules==0.2.8
pycparser==2.20
Pygments==2.7.1
PyHamcrest==2.0.2
PyJWT==1.7.1
pyOpenSSL==19.1.0
PySocks==1.7.1
pyTelegramBotAPI==3.6.6
python-dateutil==2.8.1
python-magic==0.4.18
python-memcached==1.59
python-nvd3==0.14.2
python-slugify==1.1.4
python3-openid==3.2.0
pytz==2020.1
qrcode==6.1
raven==6.10.0
redis==3.2.0
requests==2.24.0
requests-oauthlib==1.3.0
rsa==4.6
service-identity==18.1.0
simplejson==3.17.2
six==1.15.0
sqlparse==0.4.0
traitlets==5.0.4
twilio==6.24.1
Twisted==20.3.0
txaio==20.4.1
Unidecode==1.2.0
uritemplate==3.0.1
urllib3==1.25.10
vine==1.3.0
wcwidth==0.2.5
websocket-client==0.57.0
whitenoise==4.1.2
zipp==3.3.0
zope.interface==4.7.2
```
Somehow we don't need smtplib to send email through AWS SES.

Can you install a clean TSD server without manually installing smtplib? I'm pretty sure it'll send SMTP emails just fine.

drveni, 2021-04-06:
now i got this 
2021-04-06T14:51:19.725956941Z     confirmation.send(request, signup=signup),
2021-04-06T14:51:19.725960119Z   File "/usr/local/lib/python3.7/site-packages/allauth/account/models.py", line 169, in send,
2021-04-06T14:51:19.725963424Z     get_adapter(request).send_confirmation_mail(request, self, signup),
2021-04-06T14:51:19.725966634Z   File "/usr/local/lib/python3.7/site-packages/allauth/account/adapter.py", line 464, in send_confirmation_mail,
2021-04-06T14:51:19.725969880Z     self.send_mail(email_template, emailconfirmation.email_address.email, ctx),
2021-04-06T14:51:19.725974079Z   File "/usr/local/lib/python3.7/site-packages/allauth/account/adapter.py", line 136, in send_mail,
2021-04-06T14:51:19.725977488Z     msg.send(),
2021-04-06T14:51:19.725980593Z   File "/usr/local/lib/python3.7/site-packages/django/core/mail/message.py", line 291, in send,
2021-04-06T14:51:19.725986476Z     return self.get_connection(fail_silently).send_messages([self]),
2021-04-06T14:51:19.725990001Z   File "/usr/local/lib/python3.7/site-packages/django/core/mail/backends/smtp.py", line 103, in send_messages,
2021-04-06T14:51:19.725993250Z     new_conn_created = self.open(),
2021-04-06T14:51:19.725996342Z   File "/usr/local/lib/python3.7/site-packages/django/core/mail/backends/smtp.py", line 68, in open,
2021-04-06T14:51:19.725999552Z     self.connection.starttls(keyfile=self.ssl_keyfile, certfile=self.ssl_certfile),
2021-04-06T14:51:19.726002757Z   File "/usr/local/lib/python3.7/smtplib.py", line 749, in starttls,
2021-04-06T14:51:19.726006004Z     self.ehlo_or_helo_if_needed(),
2021-04-06T14:51:19.726009120Z   File "/usr/local/lib/python3.7/smtplib.py", line 600, in ehlo_or_helo_if_needed,
2021-04-06T14:51:19.726012269Z     if not (200 <= self.ehlo()[0] <= 299):,
2021-04-06T14:51:19.726015658Z   File "/usr/local/lib/python3.7/smtplib.py", line 440, in ehlo,
2021-04-06T14:51:19.726018835Z     self.putcmd(self.ehlo_msg, name or self.local_hostname),
2021-04-06T14:51:19.726022030Z   File "/usr/local/lib/python3.7/smtplib.py", line 367, in putcmd,
2021-04-06T14:51:19.726025190Z     self.send(str),
2021-04-06T14:51:19.726028204Z   File "/usr/local/lib/python3.7/smtplib.py", line 359, in send,
2021-04-06T14:51:19.726031321Z     raise SMTPServerDisconnected('please run connect() first'),
2021-04-06T14:51:19.726034434Z smtplib.SMTPServerDisconnected: please run connect() first,
2021-04-06T14:51:19.732628917Z django.channels.server ERROR    HTTP POST /accounts/email/ 500 [0.02, 192.168.5.68:61286],
2021-04-06T14:51:48.737992405Z daphne.server INFO     Killed 0 pending application instances,
2021-04-06T14:51:48.740035890Z Performing system checks...,
2021-04-06T14:51:48.740050085Z ,
2021-04-06T14:51:48.740053887Z System check identified no issues (0 silenced).,
2021-04-06T14:51:48.740057282Z April 05, 2021 - 11:10:11,
2021-04-06T14:51:48.740060543Z Django version 2.2.10, using settings 'config.settings',
2021-04-06T14:51:48.740063774Z Starting ASGI/Channels version 2.4.0 development server at http://0.0.0.0:3334/,
2021-04-06T14:51:48.740067099Z Quit the server with CONTROL-C.,
2021-04-06T14:51:54.822339559Z /usr/local/lib/python3.7/site-packages/daphne/server.py:15: UserWarning: Something has already installed a non-asyncio Twisted reactor. Attempting to uninstall it; you can fix this warning by importing daphne.server early in your codebase or finding the package that imports Twisted and importing it later on.,
2021-04-06T14:51:54.822502880Z   UserWarning,,
2021-04-06T14:51:54.822480206Z ,
2021-04-06T14:51:54.822518772Z 0 static files copied to '/app/static_build', 278 unmodified, 564 post-processed.,
2021-04-06T14:51:56.023380818Z /usr/local/lib/python3.7/site-packages/daphne/server.py:15: UserWarning: Something has already installed a non-asyncio Twisted reactor. Attempting to uninstall it; you can fix this warning by importing daphne.server early in your codebase or finding the package that imports Twisted and importing it later on.,
2021-04-06T14:51:56.023455766Z   UserWarning,,
2021-04-06T14:51:56.023431428Z Operations to perform:,
2021-04-06T14:51:56.023468045Z   Apply all migrations: account, admin, app, auth, channels_presence, contenttypes, sessions, sites, socialaccount,
2021-04-06T14:51:56.023471598Z Running migrations:,
2021-04-06T14:51:56.023474821Z   No migrations to apply.,
2021-04-06T14:51:56.023478084Z   Your models have changes that are not yet reflected in a migration, and so won't be applied.,
2021-04-06T14:51:56.023481430Z   Run 'manage.py makemigrations' to make new migrations, and then re-run 'manage.py migrate' to apply them.,
2021-04-06T14:51:57.243904680Z /usr/local/lib/python3.7/site-packages/daphne/server.py:15: UserWarning: Something has already installed a non-asyncio Twisted reactor. Attempting to uninstall it; you can fix this warning by importing daphne.server early in your codebase or finding the package that imports Twisted and importing it later on.,
2021-04-06T14:51:57.243937806Z   UserWarning,,
2021-04-06T14:51:57.243942875Z daphne.server INFO     HTTP/2 support not enabled (install the http2 and tls Twisted extras),
2021-04-06T14:51:57.244256108Z daphne.server INFO     Configuring endpoint tcp:port=3334:interface=0.0.0.0,
2021-04-06T14:51:57.244865594Z daphne.server INFO     Listening on TCP address 0.0.0.0:3334,
2021-04-06T14:52:09.730561549Z django.request ERROR    Internal Server Error: /accounts/email/,
[... cut ...]
2021-04-06T14:52:20.198607130Z django.channels.server INFO     HTTP POST /admin/app/user/?q= 302 [0.03, 192.168.5.68:61298],
2021-04-06T14:52:20.255189109Z django.channels.server INFO     HTTP GET /admin/app/user/?q= 200 [0.05, 192.168.5.68:61298],
2021-04-06T14:52:20.369207275Z django.channels.server INFO     HTTP GET /admin/jsi18n/ 200 [0.02, 192.168.5.68:61298],

sending mail from ubuntu linux server + docker(spagetti) doesn't seem to work. if this tutorial works on windows 10, im departure linux + docker, i put a new vm on server, and used on win10, and use tutorual from here. for now close/lock this issue...

drveni, 2021-04-06:
UPDATE
its works on linux server

problem is in docker-compose.yml file, i dont know correct what...

im opend portainer, and over portainer opend my docker "thespaghettidetective_web_1" then i click "edit" then i go to "Environment variables" tab 

on field

name: 
EMAIL_HOST=smtp.gmail.com
EMAIL_HOST_USER=username@gmail.com
EMAIL_HOST_PASSWORD=pass

whitout quotes or '

added new variabe
TZ=Europe/Zagreb for my correct Timezome
and it works

and new "bug"
must left under http://xxxxx:3334/admin/ on SITES if i delete server "example.com " wont up container, it freeze need reinstal
must left example.com and added new site my ip:3334
(when i recive email its putt error example.com cannot find bla bla... i dont care if he cannot find a non-existent domain )

restart docker and it works!

manual editing docker-compose.yml and put email login password wont work, whit ' or whitout '. so i did over portainer and its oke now!

thx for your lost time in this ;)

kennethjiang, 2021-04-06:
Glad you figure this out! Any idea why `TZ=xxx` fixed the problem?

drveni, 2021-04-07:
None, but in the log it's error...

Better see what's problem on docker-compose.yml, and whay it's not read this variables correctly...

I'm end of user, and i report inaccuracy

kennethjiang, 2021-04-09:
Not sure if I completely understood your point. But let me know if there is anything we can do. Otherwise I'll mark this issue closed.

closed 2022-04-26 by raymondh2