TheSpaghettiDetective/obico-server #430: buffer stream

drveni, 2021-04-18:
at home i have server in local network, gigabit network

im tried on wifi and over lan cabell same problem.

and on server have VM whit docker install and some containers
im install TSD, configured and it works.

problem is that on raspberry stream i see great on octoprint server, (octoprint linked to server on same network and subnet)
when i load TDS always "Buffering... Why is it stuck?" on octoprint.txt i put 320*480 and framerate to 10, and still wont work... sometimes i see 1 pic freezed and whit same error . on tds time-lapse i see videos normal.

on log files i see not some problems (from portainer)
when i print processor on rasppbeery is low (its rasp pi 4/8gb ram). and raspi 2 ver cam. connected whit flat cabell.

on octoprint/TSD server adress is ok, port is ok. try to disable premium webcam stream, try automatic, and always or never stream compatibility mode... try HLS or MJPG. on octoprint stream works great but TDS server live stream wont.. always buffer !!?

what im doing wrong? im updated last whit git and docker-compose...

kennethjiang, 2021-04-19:
This looks like a settings/networking issue instead of a bug (code issue). You will get help much faster by posting it in our discord server: https://discord.gg/NcZkQfj

joshbgosh10592, 2021-12-30:
I'm having the same issue, but local network isn't too bad. Off network shows me two frames and sticks on buffering.
Did you figure anything out?

drveni, 2021-12-30:
> I'm having the same issue, but local network isn't too bad. Off network shows me two frames and sticks on buffering. Did you figure anything out?

Yes, I did. The problem was that the server was too weak. I have vm on proxmox. I increased the speed of the processor and the ram. It worked normally after that.

And my network update to gigabit.

joshbgosh10592, 2021-12-30:
> Yes, I did. The problem was that the server was too weak. I have vm on proxmox. I increased the speed of the processor and the ram. It worked normally after that.
>
> And my network update to gigabit.

Wow, that's oddly enough how I have mine set up, Docker running on a Linux VM sitting on Proxmox. I have it assigned 8 vCPUs (2Ghz) though and 4GB RAM, so I figured that should be enough. My network is already virtually gigabit on the VM and 10Gb to the switch from my Proxmox host. 
I don't see the CPU usage spiking too high except for the every 5 seconds or so like their wiki says would happen. RAM seems to sit at around 83%
Glad it worked for you! I'll bump up the RAM and see if that helps at all.

drveni, 2021-12-31:
My processor is xeon at 3.4mhz. 8gb Ram. I didn't notice the increase in CPU power either. But that solved my problem.  Now I use a clipper I haven't tried the plugin for it yet

kennethjiang, 2022-01-01:
@drveni thank you for helping other TSD private server users!

Can I close this issue then?

drveni, 2022-01-01:
> @drveni thank you for helping other TSD private server users!
> 
> Can I close this issue then?

Yes... Thx!

closed 2022-01-01 by kennethjiang