TheSpaghettiDetective/obico-server #434: Failing To Run On M1 Mac mini

Alexander-T-Moss, 2021-04-19:
After installing both docker and git when I run the command, % cd TheSpaghettiDetective && docker compose up -d, I get an error and can't figure out why, my only guesses are to do with compatibility with apple silicon
[image]

kennethjiang, 2021-04-21:
I think the reason is we don't have the images for the M1 arch.

rahools, 2022-09-03:
Maybe a software update might have resolved the issue. docker-compose is working just fine for me.

OEUG99, 2022-09-04:
I've been having no issue with using [Colima](https://github.com/abiosoft/colima) and getting everything to run properly on M1 Mac Mini

nvtkaszpir, 2024-02-11:
try myabe running docker with --platform linux/amd64 on m1, it should work, but will be just slower. https://github.com/docker/for-mac/issues/6796