TheSpaghettiDetective/obico-server #442: [Feature] Notify when filament runout is detected

chrstrvs, 2021-04-24:
Hey!

Since there already is a great way to notify when spaghetti has been detected I think it would be great to also be notified when filament runout is detected and when a M600 has been issued, so that the user knows when it's time to swap filament.
If TSD can read the serial communication and listen for M600 and something to know that filament runout has been triggered (in Klipper it's very simple to add whatever action you want) then TSD can notify via e-mail, SMS or in the app.

Regards,
Christian

kennethjiang, 2021-04-25:
Thank you for suggesting a feature @chrstrvs! It's a duplicate of #392 . Can you add your comment to that ticket?

closed 2021-04-25 by kennethjiang

chrstrvs, 2021-04-25:
Sorry about that! I did a quick search for "runout" before posting, but that didn't return any results. I don't think  have much to add in #392 to what has already been written.

Keep up the great work!