TheSpaghettiDetective/obico-server #445: [BUG] TSD crashes OctoPrint, forcing restart.

charredchar, 2021-05-02:
**Describe the bug**
Created a new install of TSD Server on a Windows Server 2019 VM with Ubuntu 20.04 and following the github instructions to the T, no issues on that end. Installed the TSD plugin in OctoPrint, I was able to add my printer successfully without issue. First noticed that the camera view is not coming up on the TSD interface. During the start of the print OctoPrint will stop responding and eventually crash. This crash happens in multiple ways, either the firmware starts reporting disconnects then OctoPrint stops responding or I see no change but OctoPrint just stops responding. In all cases OctoPrint needs to be restarted from the command line. I have had this also happen without starting a print, if I leave the main page open long enough. I have also attempted all other plugins that were not required for printer operation disabled with the same results.

**Hosting environment (please complete the following information):**
TSD Server; Ubuntu 20.04 Server on a Hyper-V VM within Windows Server 2019
OctoPrint; OctoPi 0.17.0 on a Pi 4 1GB running OctoPrint 1.5.3.

Log was created on boot and ends when OctoPrint crashes.
[octoprint - Copy.log](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/files/6412045/octoprint.-.Copy.log)

charredchar, 2021-05-03:
I was working on figuring out what the cause could be, after a crash I ran dmesg, the following is the output.

[dmesg.txt](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/files/6412760/dmesg.txt)

Started watching HTOP with TSD enabled and, sure enough, as soon as I click Print it skyrockets until it fills the RAM and crashes. Without TSD enabled it'll sit under 135MBs used, even while printing. As mentioned before it can happen even if I don't start a Print, it just happens much faster if I do.

[image]

[image]

kennethjiang, 2021-05-03:
Thank you for collecting all the info. It does look like TSD plugin caused a memory leak in your Pi. It feels like it was triggered by some rare condition that happens only in small fraction of users.

The best way to troubleshoot a memory leak is to run OctoPrint with a profiler. I can't reproduce the problem otherwise I could profile it myself. Only you can do it right now since you are the only user who seem to be able to reproduce this issue. If you want to give it a shot, you could use https://docs.python.org/3/library/profile.html

charredchar, 2021-05-03:
I read over the profiler link but I am afraid that is beyond my ability. I would be happy to attempt troubleshooting but would need the help to do so to find what could cause the memory leak.

kennethjiang, 2021-05-03:
That's ok. We can try to debug it some other ways.

Can you switch to using TSD cloud and see if the memory leak still happens? 

Also I noticed that gstreamer failed to start in your Pi. This happens to some users and it's usually not a problem because the plugin would just automatically switch to compatibility mode. But I'm not sure if it'd be the cause. Can you let me know if you are getting premium webcam streaming once you switch to TSD cloud?

charredchar, 2021-05-03:
I have premium disabled as it causes issues with other plugins such as OctoLapse, I won't be monitoring the printer via the TSD interface so premium is not needed in my case.

I would be willing to test using the cloud service as well, though I will need to make an account later today for testing. I have only ever used TSD with a private server and only intend to.

kennethjiang, 2021-05-03:
When did you disable the premium streaming? Has the memory leak happened since you disabled it? Your log file shows that the premium streaming was still enabled, can you confirm that?

charredchar, 2021-05-03:
I disable it as soon as I am able to get into the settings to disable it. Since it defaults to on the log reflects a new install, it still happens after disabling premium. I thought I might have disabled it during that log as well but maybe it did not? I can check on this as well, install, disable premium and reboot to check the log and ensure it isn't enabled. What should I look for in the log to tell if it is enabled or disabled?

kennethjiang, 2021-05-03:
In your log, you have `'disable_video_streaming': False`. It should be `'disable_video_streaming': True`

charredchar, 2021-05-03:
Thanks! Later on tonight I'll be able to run those tests and report back.

charredchar, 2021-05-04:
Okay, so I was able to tinker a little and came across a couple of things.
First, reinstalled the plugin, did not start seeing the memory leak until a minute or two after setup. I was able to disable premium streaming and restart OctoPrint before it crashed due to lack of memory. Once OctoPrint restarted I reloaded the page and the memory leak did not re-emerge until after I started a print. So it seems two things cause it, Premium Streaming being enabled or while printing. I attempted to cancel the print to see if the leak would halt but my start macro would have taken too long and crashed, I restarted OctoPrint to prevent the crash.
Second thing I noticed, this whole time the status in TSD is switching back and forth between "Operational" (or Canceling, Printing, etc) and Offline. The stream, even a snapshot version, is not showing up in TSD and it only shows loading icon with "Buffering."
I will attempt the cloud service shortly, in the mean time I wanted to upload an updated log of this recent attempt.

[octoprint2 - Copy.log](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/files/6418805/octoprint2.-.Copy.log)

charredchar, 2021-05-04:
I attempted with TSD Cloud, exactly the same where it will leak while premium is enabled or printing. I don't see the printer switching states on the TSD interface though like it does with my local server.

Attached is the log from the point the previous one leaves off. Please let me know if there is anything else I can test.

[octoprint3 - Copy.log](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/files/6418844/octoprint3.-.Copy.log)

kennethjiang, 2021-05-04:
Can't see anything abnormal from the log file.

One thing strange is both janus and gstreamer failed to start, apparently from some compatibility problem of the native libs. Have you upgraded the native libs (such as running "apt-get upgrade") or running it on top of docker?

charredchar, 2021-05-04:
I have not ran it beyond the initial install of the system, both on the Ubuntu server or OctoPi. I can do it on both this afternoon and report back.

kennethjiang, 2021-05-04:
Don't run "apt-get upgrade" or docker. Please stay with the original octopi to minimize the change of errors.

If you have a spare SD card, please flash it with OctoPi 0.18 and let me know if the memory leak still happens.

charredchar, 2021-05-04:
A spare SD with 0.18 is going to be a little harder as I run Klipper and the firmware is locked to the Klipper install I currently have running on 0.17. I plan to update Klipper shortly and can test then, but either way it won't be for a couple of weeks.

I do have an image of stock 0.17 after the fresh install of Klipper before all else I can also test TSD with. This image is basically the 0.17 installed with Klipper and nothing else so it could be before any other plugins caused an issue. This will need to wait until the weekend though as well as I am away from home and been doing the testing remotely.

kennethjiang, 2021-05-04:
Sounds good!

kennethjiang, 2021-05-07:
@charredchar can you confirm you are on OctoPrint 1.5.3, not 1.6.0? Another user ran into a similar problem on 1.6.0, and got the problem solved after rolling it back to 1.5.3.

closed 2022-04-26 by raymondh2