TheSpaghettiDetective/obico-server #456: [BUG] Jetson Nano 4GB TSD Web Server not starting.

hendrikreuhl, 2021-05-20:
**Describe the bug**
I followed the guide from https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/docs/jetson_guide.md and tried to install TSD on my Jetson. Unfortunately, the web server does not start

I googled around a lot and searched the repo issues and looked at the discord, but was not able to resolve the issue by myself.

Here you can find the log attached:
`ModuleNotFoundError: No module named 'corsheaders'
/usr/local/lib/python3.7/site-packages/daphne/server.py:15: UserWarning: Something has already installed a non-asyncio Twisted reactor. Attempting to uninstall it; you can fix this warning by importing daphne.server early in your codebase or finding the package that imports Twisted and importing it later on.
  UserWarning,
Traceback (most recent call last):
  File "manage.py", line 15, in <module>
    execute_from_command_line(sys.argv)
  File "/usr/local/lib/python3.7/site-packages/django/core/management/__init__.py", line 381, in execute_from_command_line
    utility.execute()
  File "/usr/local/lib/python3.7/site-packages/django/core/management/__init__.py", line 357, in execute
    django.setup()
  File "/usr/local/lib/python3.7/site-packages/django/__init__.py", line 24, in setup
    apps.populate(settings.INSTALLED_APPS)
  File "/usr/local/lib/python3.7/site-packages/django/apps/registry.py", line 91, in populate
    app_config = AppConfig.create(entry)
  File "/usr/local/lib/python3.7/site-packages/django/apps/config.py", line 90, in create
    module = import_module(entry)
  File "/usr/local/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked`

kennethjiang, 2021-05-21:
Is this the same issue as #424 ?

LyricPants66133, 2021-06-07:
@hendrikreuhl this issue has been fixed. @kennethjiang you can resolve this issue.

kennethjiang, 2021-06-07:
@hendrikreuhl You need to pull from the repo and follow the revised jetson guide. Let me know if it still doesn't work for you.

closed 2021-06-07 by kennethjiang