TheSpaghettiDetective/obico-server #458: [HELP REQUEST] WebRTC : ICE failed, your TURN server appears to be broken

B-0-B-B-Y, 2021-05-24:
# Environment Setup

### The Spaghetti Detective

Running inside Docker v20.10.6 on a Windows 10 LTSC PC running WSL1 Debian. In windows defender firewall, I have created a rule for port 3334 to be open to the LAN.

### OctoPrint

Running on a RasPi 4 using the official Octoprint RasPi image. Also connected to the LAN network. TheSpaghettiDetective plugin is installed and configured with the local spaghetti detective server running inside docker on the Windows 10 LTSC PC. 

# Description of Issue

My issue revolves around not being able to see the webcam inside The Spaghetti Detective, which means the main feature (AI watching my prints) is not possible. I have followed the suggested troubleshooting tips of waiting 30 seconds, starting a print and seeing if the video feed comes through, performing the test connection inside the OctoPrint interface and so on. None of those have resolved the issue.

I can confirm that everything else works as I am able to move the printer X/Y/Z, set bed/extruder temps, tunnel into the OctoPrint interface successfully etc.

In the browser, when on the TSD web interface, opening the dev console I can see a few errors, here's a dump of said logs:

```
 ::: Got a remote stream ::: webrtc.js:107:22
MediaStream { id: "janus", active: true, onaddtrack: null, onremovetrack: null }
webrtc.js:108:22
Adding onended callback to track: 
MediaStreamTrack { kind: "video", id: "{8dd8aebb-9b01-42d8-82f4-b84b590ff5d7}", label: "remote video", enabled: true, muted: true, onmute: onmute(t), onunmute: onunmute(e), readyState: "live", onended: onended(e)
 }
janus.js:1800:14
Remote description accepted! janus.js:1873:16
Creating answer (iceDone=false) janus.js:2695:12
isAudioSendEnabled: 
Object { audioSend: false, videoSend: false, data: true, update: false, keepAudio: false, keepVideo: false }
janus.js:3422:10
isAudioRecvEnabled: 
Object { audioSend: false, videoSend: false, data: true, update: false, keepAudio: false, keepVideo: false }
janus.js:3444:10
Adding recvonly audio transceiver: 
RTCRtpTransceiver { mid: null, sender: RTCRtpSender, receiver: RTCRtpReceiver, stopped: false, direction: "recvonly", currentDirection: null }
janus.js:2782:18
isVideoSendEnabled: 
Object { audioSend: false, videoSend: false, data: true, update: false, keepAudio: false, keepVideo: false }
janus.js:3455:10
isVideoRecvEnabled: 
Object { audioSend: false, videoSend: false, data: true, update: false, keepAudio: false, keepVideo: false }
janus.js:3477:10
Setting video transceiver to recvonly: 
RTCRtpTransceiver { mid: "video", sender: RTCRtpSender, receiver: RTCRtpReceiver, stopped: false, direction: "recvonly", currentDirection: "recvonly" }
janus.js:2839:20
Object { … }
janus.js:2865:10
isVideoSendEnabled: 
Object { audioSend: false, videoSend: false, data: true, update: false, keepAudio: false, keepVideo: false }
janus.js:3455:10
Object { type: "answer", sdp: "v=0\r\no=mozilla...THIS_IS_SDPARTA-88.0.1 2111081279350468754 0 IN IP4 0.0.0.0\r\ns=-\r\nt=0 0\r\na=fingerprint:sha-256 84:54:31:3A:8B:3A:51:A6:43:98:5A:2F:A3:70:F4:BA:01:1E:74:B7:B3:6A:23:2A:C1:89:AE:46:02:2D:B5:B6\r\na=group:BUNDLE video data\r\na=ice-options:trickle\r\na=msid-semantic:WMS *\r\nm=video 9 UDP/TLS/RTP/SAVPF 96 97\r\nc=IN IP4 0.0.0.0\r\na=recvonly\r\na=fmtp:96 profile-level-id=42e01f;level-asymmetry-allowed=1;packetization-mode=1\r\na=fmtp:97 apt=96\r\na=ice-pwd:38d14eddd83fdac87b9a9c8fb4251060\r\na=ice-ufrag:dd965967\r\na=mid:video\r\na=rtcp-fb:96 nack\r\na=rtcp-fb:96 nack pli\r\na=rtcp-fb:96 goog-remb\r\na=rtcp-mux\r\na=rtpmap:96 H264/90000\r\na=rtpmap:97 rtx/90000\r\na=setup:active\r\na=ssrc:3866016593 cname:{8295d420-1903-4a55-9c29-7e7a5f8498a6}\r\nm=application 9 DTLS/SCTP 5000\r\nc=IN IP4 0.0.0.0\r\na=sendrecv\r\na=ice-pwd:38d14eddd83fdac87b9a9c8fb4251060\r\na=ice-ufrag:dd965967\r\na=mid:data\r\na=sctpmap:5000 webrtc-datachannel 256\r\na=setup:active\r\na=max-message-size:1073741823\r\n" }
janus.js:2885:14
Setting local description janus.js:2894:14
Got SDP! webrtc.js:183:20
Object { type: "answer", sdp: "v=0\r\no=mozilla...THIS_IS_SDPARTA-88.0.1 2111081279350468754 0 IN IP4 0.0.0.0\r\ns=-\r\nt=0 0\r\na=fingerprint:sha-256 84:54:31:3A:8B:3A:51:A6:43:98:5A:2F:A3:70:F4:BA:01:1E:74:B7:B3:6A:23:2A:C1:89:AE:46:02:2D:B5:B6\r\na=group:BUNDLE video data\r\na=ice-options:trickle\r\na=msid-semantic:WMS *\r\nm=video 9 UDP/TLS/RTP/SAVPF 96 97\r\nc=IN IP4 0.0.0.0\r\na=recvonly\r\na=fmtp:96 profile-level-id=42e01f;level-asymmetry-allowed=1;packetization-mode=1\r\na=fmtp:97 apt=96\r\na=ice-pwd:38d14eddd83fdac87b9a9c8fb4251060\r\na=ice-ufrag:dd965967\r\na=mid:video\r\na=rtcp-fb:96 nack\r\na=rtcp-fb:96 nack pli\r\na=rtcp-fb:96 goog-remb\r\na=rtcp-mux\r\na=rtpmap:96 H264/90000\r\na=rtpmap:97 rtx/90000\r\na=setup:active\r\na=ssrc:3866016593 cname:{8295d420-1903-4a55-9c29-7e7a5f8498a6}\r\nm=application 9 DTLS/SCTP 5000\r\nc=IN IP4 0.0.0.0\r\na=sendrecv\r\na=ice-pwd:38d14eddd83fdac87b9a9c8fb4251060\r\na=ice-ufrag:dd965967\r\na=mid:data\r\na=sctpmap:5000 webrtc-datachannel 256\r\na=setup:active\r\na=max-message-size:1073741823\r\n" }
webrtc.js:184:20
Sending message to plugin (handle=3115088913008970): janus.js:1305:10
Object { janus: "message", body: {…}, transaction: "3RAsk4yQ0QOE", jsep: {…}, session_id: 5151605288651453, handle_id: 3115088913008970 }
janus.js:1306:10
Sending trickle candidate (handle=3115088913008970): janus.js:1401:10
Object { janus: "trickle", candidate: {…}, transaction: "XNwvi2yz0LLu", session_id: 5151605288651453, handle_id: 3115088913008970 }
janus.js:1402:10
Got an ack on session 5151605288651453 janus.js:611:12
Object { janus: "ack", session_id: 5151605288651453, transaction: "3RAsk4yQ0QOE" }
janus.js:612:12
Message sent! janus.js:1311:14
Object { janus: "ack", session_id: 5151605288651453, transaction: "3RAsk4yQ0QOE" }
janus.js:1312:14
Got a plugin event on session 5151605288651453 janus.js:762:12
Object { janus: "event", session_id: 5151605288651453, transaction: "3RAsk4yQ0QOE", sender: 3115088913008970, plugindata: {…} }
janus.js:763:12
  -- Event is coming from 3115088913008970 (janus.plugin.streaming) janus.js:774:12
Object { streaming: "event", result: {…} }
janus.js:776:12
Notifying application... janus.js:789:14
 ::: Got a message ::: webrtc.js:154:12
Object { streaming: "event", result: {…} }
webrtc.js:155:12
Starting webrtc.js:161:20
Got an ack on session 5151605288651453 janus.js:611:12
Object { janus: "ack", session_id: 5151605288651453, transaction: "XNwvi2yz0LLu" }
janus.js:612:12
Starting bitrate timer (via getStats) janus.js:3082:14
0 kbits/sec webrtc.js:96:28
Received state change on data channel: 
close { target: RTCDataChannel, isTrusted: true, srcElement: RTCDataChannel, currentTarget: RTCDataChannel, eventPhase: 2, bubbles: false, cancelable: false, returnValue: true, defaultPrevented: false, composed: false, … }
janus.js:1447:12
WebRTC: ICE failed, your TURN server appears to be broken, see about:webrtc for more details
State change on <JanusDataChannel> data channel: closed janus.js:1450:12
Got an ack on session 5151605288651453 janus.js:611:12
Object { janus: "ack", session_id: 5151605288651453, transaction: "BHtLpNoP31Fz" }
janus.js:612:12
0 kbits/sec webrtc.js:96:28
Sending trickle candidate (handle=3115088913008970): janus.js:1401:10
Object { janus: "trickle", candidate: {…}, transaction: "DpX5PteEHKCu" }
janus.js:1402:10
End of candidates. janus.js:1768:16
Sending trickle candidate (handle=3115088913008970): janus.js:1401:10
Object { janus: "trickle", candidate: {…}, transaction: "Mxs1xhia67Zx" }
janus.js:1402:10
Got an ack on session 5151605288651453 janus.js:611:12
Object { janus: "ack", session_id: 5151605288651453, transaction: "DpX5PteEHKCu" }
janus.js:612:12
Got an ack on session 5151605288651453 janus.js:611:12
Object { janus: "ack", session_id: 5151605288651453, transaction: "Mxs1xhia67Zx" }
janus.js:612:12
0 kbits/sec
```

Is anyone able to help me out please, I don't have the most experience with WebRTC so am not sure how to rectify this issue. As a note, I get the `ICE failed, your TURN server appears to be broken` message when using the official TSD server as well and cannot see the webcam stream when the printer is idling, I can only see the webcam when actually printing, not sure if that is helpful or not but wanted to mention it regardless. Locally the camera never works, whether idling or printing.

Thank you in advance! 😃

B-0-B-B-Y, 2021-05-24:
UPDATE: So this was a pretty DOH moment, as is usually the case, I needed to add my LAN IP as a new site inside of the django admin panel... so adding `my-lan-ip:3334` to the sites configurator, I could see my camera 👍 only trouble now is, it's very choppy / constantly buffering, I have a 2.5GBs LAN card that I use for my PC that's running docker and the Raspi has gigabit ethernet afaik, so there should be no issue streaming from the Raspi Cam v2.0 hooked up to the Raspi. The camera was smooth when I was using the official cloud hosted TSD server, so not sure why locally it would perform worse? Docker is set to 4 CPU cores (CPU is a 5950X) and 12GB of RAM (PC has 64GB of RAM) so not sure what is causing this video buffering lag, if anyone has any ideas?

B-0-B-B-Y, 2021-05-24:
UPDATE 2: So the lagging streaming seems to be browser specific. Opening the local TSD server in the latest Chrome / Edge (as of this post) and the mobile version of Firefox, the video feed is crisp and smooth, no buffering or dropped frames. In Firefox v88.0.1 (64-bit) however, the video issue described above is observed. Will look at the frontend code a bit later, try and do some debugging, if I have a fix for the Firefox player behaviour, will open a PR and reference this issue 👍 From the initial logs, it looks like the bit rate never goes above 0kbps and instead of streaming video it is streaming single frames at every X interval (haven't looked close enough at what the X interval is in actual seconds)

kennethjiang, 2021-05-24:
Sorry for the delay and good to see you have done a lot of digging yourself.

TSD cloud runs a TURN server that only serves TSD Pro subscribers in the cloud. That's probably why you saw that error. However, 2 things to point out:

1. WebRTC is for premium streaming only. You don't need it for failure detection. https://help.thespaghettidetective.com/kb/guide/en/premium-streaming-versus-basic-streaming-QHLFp22YVM/Steps/297219
2. TURN server is optional. Most of the time WebRTC works on the same LAN without TURN. But we have seen a lot of exceptions to that.

From what you described, the WebRTC connection is already created (without TURN). However, somehow the connection has a bottleneck somewhere, most likely with the wifi connection of your Pi.

Since this is an environment/setup issue, not a code issue, you will get help much faster in our discord server: https://discord.gg/NcZkQfj

B-0-B-B-Y, 2021-05-26:
@kennethjiang Thanks for getting back to me and explaining more about the internal setup 👍 I've ruled out a connection issue, as testing from different browsers and my mobile phone, all video feeds work at smooth FPS, with the exception of desktop firefox. I did some more debugging and it looks like when viewing the client from the desktop firefox browser, it doesn't look like it's hitting the `@canplay` event in `StreamingBox.vue` in my desktop firefox. Not a deal breaker or anything like that, as like I said, can use my phone / other browser to watch the stream, just would be nice to fix. I can try and do some code changes in my local images, and if I find something that works, I can open a PR for it, but am busy with my day job over the week so probably won't happen until the weekend.

Thanks for the discord link, I'll message people around there if my debugging over the weekend doesn't provide anything fruitful

Thanks for this cool project regardless, I was wondering, for the mobile app that you have, is the source code for it also open source or are you keeping it closed source? Just asking because I couldn't find a feature in the mobile app that let me connect it to my own local server, not sure if that is already a feature and if not, I would be interested in contributing, I can't imagine it would take much effort to allow for that kind of functionality (assuming it's open source and you're happy with that kind of feature being introduced of course) 🙂

kennethjiang, 2021-05-27:
Great. Looking forward to your PR.

The mobile app doesn't work with the private server yet. We have a task open for that: https://trello.com/c/PJmB2NS0/178-work-with-private-tsd-server

Hopefully we can get to that task in the next few weeks. But open-sourcing the mobile app will take much longer, as the mobile app needs a lot of credentials (the ones required by the app stores, the keys to allow social sign-in, etc) and there are no established mechanisms to keep those credentials outside the code. We decided to put all credentials in the source code so that we could push the app out faster. And it'll probably take a long time before we can figure out (if we ever could) a way to extract all of them out of the source code.

closed 2022-04-26 by raymondh2