TheSpaghettiDetective/obico-server #465: [Feature] Ability to connect Android/iOS app to personally hosted server

joshbgosh10592, 2021-06-05:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I know of quite a few people that would like the ability to connect the Android/iOS app to a personally hosted server. I am one of them. Please?

**Describe alternatives you've considered**
Using a mobile web browser instead, but that doesn't give push notifications, or a persistent notification.

**Additional context**
Please?

LyricPants66133, 2021-06-07:
@joshbgosh10592 This has been discussed a lot on the discord, and is impossible to do at the current moment due to a variety of reasons. You do have alternatives however like a discord bot plugin for octoprint or a webhook for TSD. Sorry to disappoint.

@kennethjiang this issue can be resolved.

kennethjiang, 2021-06-07:
For whatever reason, I missed this issue.

We actually want to make the app compatible with the private server. We have been tied up by a bunch of other things that have popped up and taken priority.

Let's keep this issue open until we got a chance to really make it happen.

kennethjiang, 2021-11-27:
We have an option to do it now. Go to "About" -> "Developer Options".

closed 2021-11-27 by kennethjiang

joshbgosh10592, 2021-12-30:
You are fantastic. Thank you!!!