TheSpaghettiDetective/obico-server #472: [BUG] Greyscale meter with overlay

kennethjiang, 2021-06-10:
[image]

dipuzyrev, 2021-06-11:
@kennethjiang this issue requires colors adjustment, right?

dipuzyrev, 2021-06-11:
BTW, any dynamic color have to be in `themes.js` file. I'll just leave a comment these colors are one-offs

dipuzyrev, 2021-06-11:
Here is `#2d3e4f` color pointed as page background, but page background is `#23313f` (in dark mode, regarding to color guide). Where is the truth?)
[image]

dipuzyrev, 2021-06-11:
And are you sure about making gauge background same as page background. It will look like this [image]

dipuzyrev, 2021-06-11:
This is what I have now. First part — how it looks now in light and dark version, second — how it would look if we make gague background same as page.

https://user-images.githubusercontent.com/12683472/121705970-da100c00-cadd-11eb-853c-8ed4266b8c40.mov

closed 2021-06-11 by kennethjiang