TheSpaghettiDetective/obico-server #493: [BUG] Docker images using outdated version of Python

lps-rocks, 2021-07-05:
**Describe the bug**
When building the docker containers the current warning is thrown by pip when being installed / upgraded. This could lead to potential security issues since the underlying interpreter version is not getting updates any longer.

```DEPRECATION: Python 3.5 reached the end of its life on September 13th, 2020. Please upgrade your Python as Python 3.5 is no longer maintained. pip 21.0 will drop support for Python 3.5 in January 2021. pip 21.0 will remove support for this functionality.```

**To Reproduce**
Steps to reproduce the behavior:
1. `git clone https://github.com/TheSpaghettiDetective/TheSpaghettiDetective.git
2. `cd TheSpaghettiDetective && sudo docker-compose up -d`
3. watch build

**Hosting environment (please complete the following information):**
 - OS: Debian Linux
 - Version: 10.10
 - Hardware: x86_64 PC

**Additional context**

kennethjiang, 2021-07-07:
We have tried to upgrade to 3.7 but it turned out to be trickier than we thought. So we had to roll it back assuming our effort can be better spent than chasing upgrades.

This deprecation is really just a warning. Nothing will be broken. I will keep this ticket open as we will circle back to it at some point. But it won't be a priority soon. Meanwhile, anyone who wants to take a stab at it please do so.

lps-rocks, 2021-07-07:
Are there any records of what went wrong trying to move to 3.7? Might give the person taking the stab a head start :)

kennethjiang, 2021-07-07:
Unfortunately we don't have a record for that. Also this PR may potentially make the upgrade a lot easier: https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/pull/492. Please hold off on this issue until #492 is merged.

encetamasb, 2021-07-14:
Hi, I would like to clarify: it's the ml_api image what still uses python 3.5. 
Other services are on python 3.7, the referred PR does not affect ml_api.

jsdokos, 2021-10-29:
I recieved this warning when installing on Ubuntu today. Looks like updating Python should be a priority.

`DEPRECATION: Python 3.5 reached the end of its life on September 13th, 2020. Please upgrade your Python as Python 3.5 is no longer maintained. pip 21.0 will drop support for Python 3.5 in January 2021. pip 21.0 will remove support for this functionality.`