TheSpaghettiDetective/obico-server #496: [Feature] Alert when printer pauses

Doormat1, 2021-07-08:
If you have action host and emergency parser enabled in firmware, when the printer pauses when the BTT smart filament sensor is triggered, it also pauses octoprint. It would be great if TSD could alert us of the pause.

mrx23dot, 2021-09-15:
Great idea, detecting no motion, while ignoring ambient light.

kennethjiang, 2021-09-16:
Will OctoPrint send an event to the plugin in such case? If so, it'd be a very trivial feature to add. @mrx23dot @Doormat1

Doormat1, 2021-09-16:
I'm not sure, Octoprint does have a pop up when the filament sensor is
tripped and the TSD does recognise that the print is paused

mrx23dot, 2021-09-16:
You could just wire in to the current output, saying attention needed, I wouldnt stop the printing until it gets bulletproof.
(eg. during heat up nothing moves)