TheSpaghettiDetective/obico-server #508: [Feature] Detect filament not extruding

kunalgrover05, 2021-08-23:
**Is your feature request related to a problem? Please describe.**
On the stock Ender, I saw a lot of issues where filament stopped extruding. Those issues are solved with me upgrading to a better extruder but they still happen occasionally when printing filaments like TPU

**Describe the solution you'd like**
TSD is at the right place to detect and pause prints if the extruder is not extruding filament. It might not work very well with a camera at an inclination but will work great at cameras which can see the nozzle.

**Describe alternatives you've considered**
Optical filament sensors don't work very well when the filament is jammed in the extruder stepper / Bowden setup. 

**Additional context**
NA

kunalgrover05, 2021-09-06:
There are 2 kinds of sensors in the market today:
- **Filament runout sensor**: Optical sensors which only detect whether the filament has run-out to pause printing. I believe these are best handled by the sensor. 
- **Smart Filament detection sensor**: These sensors do what Filament runout sensors do, but beyond that also work for detecting if filament is not extruding. eg Blocked nozzle, Flexible Filament stuck etc. 

I think TSD can lie in the second bucket where it can detect all kinds of reasons for filament not extruding and pause the print.

kennethjiang, 2021-09-06:
There was a good discussion in TSD's forum recently about this topic: https://discord.com/channels/614543405724205137/661327037360898063/881298295778082866

Please take a look and let me know if you don't think the sensors proposed in the forum will work for you. They would be a much easier and much faster way compared to using AI.

mrx23dot, 2021-09-15:
Stall sensors are not easy to calibrate, because optimally you need extruder input to compare filament feed, otherwise if you wait long for heat up it false triggers.

It should be possible to detect gap between camera and model, if you have a white background, but if you print with white you are out of luck, also the camera has to be on same height as nozzle to move alone.

Maybe the easiest is detecting stall for 5minutes, and checking on printer every 1hr.

mrx23dot, 2021-09-15:
Here is a stall sensor that compares filament demand-supply, optically, not cheap.
https://www.tindie.com/products/niujl123/laser-filament-motion-sensor/

kennethjiang, 2021-09-16:
@mrx23dot Thank you so much for the info! I don't have a Pursa, but I have heard Prusa can detect jammed filament. The info I found only seem to be conflicting between vouching for this claim and discrediting it. Do you have more insights about Prusa's sensor?

mrx23dot, 2021-09-16:
Prusa as I know had an optical only presence detector but it wasnt reliable.
They switched to roller+lever+IR gate (same as switch). 
https://help.prusa3d.com/en/article/ir-filament-sensor-troubleshooting-mk2-5s-mk3s_112226/
similar to https://www.thingiverse.com/thing:3063430
None of them are smart enough to consider filament demand.
The only one that I have seen is the link above.

kennethjiang, 2021-09-21:
Great info. Thanks again @mrx23dot .