TheSpaghettiDetective/obico-server #510: [BUG] Verification code wrong on all printers

svdstoep, 2021-08-31:
Since some time i'm having issues with my connected printers. They aren't connected to TSD anymore. I've removed them all from the TSD config and tried adding them back in. Before I tried this i've removed the application from Octoprint and reinstalled it.  But I still can't link them.

On every printer it stops at the verification, when entering the code it says; 

Wrong code!
Did you enter an expired code?

The plugin version is;  1.8.0

Server version I can't find, but I recently upgraded it with the commands below

cd TheSpaghettiDetective
git pull
sudo docker-compose up --build -d

encetamasb, 2021-08-31:
@svdstoep Hi! http://xxxx.xxxx.xxxxx:3334/api/v1/octo/unlinked/ url is not needed for manual linking (when you type in the code). If you see this error in octoprint log, please check the ip:port pair of your server from your pi's shell.

svdstoep, 2021-08-31:
The part I entered in the Plugin configuration is: http://xxxx.xxxx.xxxxx:3334/, it seems that I can talk the the TSD server because it say's 'Wrong Code'. If I enter an incorrect host it says Failed to talk to The Spaghetti Detective server!

GIT LOG -1 gives the result you see below
[image]

encetamasb, 2021-08-31:
> The part I entered in the Plugin configuration is: http://xxxx.xxxx.xxxxx:3334/, it seems that I can talk the the TSD server because it say's 'Wrong Code'. If I enter an incorrect host it says Failed to talk to The Spaghetti Detective server!

Yes, you are right.

> GIT LOG -1 gives the result you see below
> [image]

This is a rather old version, from January. You need a quite recent one, what's compatible with 1.8.0 plugin version.

Pull it again and check what's wrong.

kennethjiang, 2021-09-01:
@svdstoep Here is how you can upgrade the server: https://github.com/TheSpaghettiDetective/TheSpaghettiDetective#upgrade-server

svdstoep, 2021-09-04:
Thanks! That was the problem, I think i've done the upgrade wrong the first time.

closed 2021-09-04 by svdstoep