TheSpaghettiDetective/obico-server #515: Jetson Nano: CUDA driver version is insufficient for CUDA runtime version

jonwilliams84, 2021-09-14:
``Running on a Jetson Nano and it appears to be working, however I am getting this spammed in the ml_api container logs;

```
/usr/bin/python3: [2021-09-14 09:21:53 +0000] [87] [INFO] Booting worker with pid: 87
layer     filters    size              input                output
   0  Try to set subdivisions=64 in your cfg-file. 
 Try to load cfg: /app/model/model.cfg, weights: /app/model/model.weights, clear = 0 
CUDA status Error: file: ./src/cuda.c : () : line: 203 : build time: Oct 15 2019 - 16:00:39 
CUDA Error: CUDA driver version is insufficient for CUDA runtime version
```

Running device query this is the result:
```
 CUDA Device Query (Runtime API) version (CUDART static linking)

Detected 1 CUDA Capable device(s)

Device 0: "NVIDIA Tegra X1"
  CUDA Driver Version / Runtime Version          10.2 / 10.0
  CUDA Capability Major/Minor version number:    5.3
  Total amount of global memory:                 3964 MBytes (4156719104 bytes)
  ( 1) Multiprocessors, (128) CUDA Cores/MP:     128 CUDA Cores
  GPU Max Clock rate:                            922 MHz (0.92 GHz)
  Memory Clock rate:                             13 Mhz
  Memory Bus Width:                              64-bit
  L2 Cache Size:                                 262144 bytes
  Maximum Texture Dimension Size (x,y,z)         1D=(65536), 2D=(65536, 65536), 3D=(4096, 4096, 4096)
  Maximum Layered 1D Texture Size, (num) layers  1D=(16384), 2048 layers
  Maximum Layered 2D Texture Size, (num) layers  2D=(16384, 16384), 2048 layers
  Total amount of constant memory:               65536 bytes
  Total amount of shared memory per block:       49152 bytes
  Total number of registers available per block: 32768
  Warp size:                                     32
  Maximum number of threads per multiprocessor:  2048
  Maximum number of threads per block:           1024
  Max dimension size of a thread block (x,y,z): (1024, 1024, 64)
  Max dimension size of a grid size    (x,y,z): (2147483647, 65535, 65535)
  Maximum memory pitch:                          2147483647 bytes
  Texture alignment:                             512 bytes
  Concurrent copy and kernel execution:          Yes with 1 copy engine(s)
  Run time limit on kernels:                     Yes
  Integrated GPU sharing Host Memory:            Yes
  Support host page-locked memory mapping:       Yes
  Alignment requirement for Surfaces:            Yes
  Device has ECC support:                        Disabled
  Device supports Unified Addressing (UVA):      Yes
  Device supports Compute Preemption:            No
  Supports Cooperative Kernel Launch:            No
  Supports MultiDevice Co-op Kernel Launch:      No
  Device PCI Domain ID / Bus ID / location ID:   0 / 0 / 0
  Compute Mode:
     < Default (multiple host threads can use ::cudaSetDevice() with device simultaneously) >

deviceQuery, CUDA Driver = CUDART, CUDA Driver Version = 10.2, CUDA Runtime Version = 10.0, NumDevs = 1
Result = PASS
```

Any ideas?

kennethjiang, 2021-09-14:
It looks like something is broken. Busy this week but may be able to find a time to look into this issue this weekend or next week.

In the mean time, you may get help faster by asking in our discord server: https://discord.com/invite/NcZkQfj

jonwilliams84, 2021-09-15:
I worked out that it was the JetPack image I was using. I had updated to 4.6 from 4.2.1.

Consequently CUDA 10.2 was installed and the images + app is built for CUDA 10.

I have resolved for now by dropping back to JetPack 4.2.1.

kennethjiang, 2021-09-16:
That'd explain why. CUDA is notoriously known for its backward compatibility (or the lack of it).

closed 2021-09-16 by kennethjiang