TheSpaghettiDetective/obico-server #516: [Feature] Detect Smoke/Fire

mrx23dot, 2021-09-17:
We could have a 2nd AI run on the image to detect safety concern and give (any) alerts.

eg. hotend starts to glow red, or the whole thing is already in flames and generates smoke.

Just an idea to push boundaries.

kennethjiang, 2021-09-18:
Good point. We definitely have thought about it.

The biggest problem we have is training data. Training an AI depends on a lot of training data, including both the positive data (smoke/fire is present in the camera) and negative data. Negative data is easy to get but positive data is not.

Any idea how we can obtain a lot of positive data? We can't scrape the web with any picture that contains fire/smoke, as the training data needs to be realistic. Pictures of camp fire, for instance, won't be helpful here... :(

mrx23dot, 2021-09-20:
We could easily do smoke (vape), and glowing of hot end (with orange LED).

Fire might be already to late although controllers can catch fire, 
but a quick hack would be to overlay real fire onto a normal print footage by video editing.
Randomly resizing them, in random rotation, brightness. 

It might be smart to do some pre-filtering/data extraction before AI, like black and white image based on orange color only, if number of orange is increasing that could be a sign.

Found some existing ones, they could work without any training:
https://www.youtube.com/watch?v=ts3yxfNrDnY
https://github.com/gengyanlei/fire-smoke-detect-yolov4
https://link.springer.com/content/pdf/10.1007/s11554-020-01044-0.pdf
https://github.com/Project-X-UBC/smoke-detection-net
https://github.com/robmarkcole/fire-detection-from-images
https://www.youtube.com/watch?v=mpw-oIvjB70
https://github.com/TCHsieh/EarlyFireDetection

kennethjiang, 2021-09-20:
Thanks again for the ideas.

A big problem with AI is once the training data is "augmented", instead of from the real scenarios, the accuracy suffers big time. Especially if the raw pictures to augment the training data are not from the real scenarios, it's also impossible to train an AI model that's good enough.

Another problem is the training data needs to be "diverse" too. For instance, if the bulk of the smoke training data are the pictures of vapor steam from a very limited number of users, the AI model will be very good at detecting vapor steam for these users, but nothing else. 

Just some of the things I learned from pouring hundreds of hours in training the AI model for TSD. :)

mrx23dot, 2021-09-21:
Then it might worth trying off the shelf already trained open source models like above.

lnjustin, 2023-01-06:
+1 for this. Would be great if the challenges on AI training could be solved and this feature provided. Talk about value add and marketing possibilities!

puterboy, 2023-12-21:
I find it's easier and probably far more accurate and robust to do the following:
- Throw a cheap smoke detector in your printer enclosure (I got mine for $10 or pay more for a networked one tied into your home fire alarm monitoring system)
- Add a cheap i2c temperature sensor to your Raspberry PI and use one of the Octoprint plugins together with a SmartPlug to shut off power to the Printer when temperature exceeds a threshhold (or use an independent script running on your PI or other Home Automation device)

That way I both *detect* and *act* on smoke/fire.

Seems much more robust than relying on AI (with all the challenges of training noted above) plus ensuring that all the elements of Obico are running properly and have network connectivity.

I don't want to have only say a 90% chance of detecting a fire -- and even spaghetti detection after years of training is far from perfect and detection is often delayed until there is enough mass to detect it.

Smoke detectors and temperature sensors are simple and robust, having been tested for decades across billions of devices.

AI is cool and powerful but just because you have a hammer, doesn't mean everything you see is a nail...