TheSpaghettiDetective/obico-server #519: [BUG] /ent/subscription/ not found

TempestWales, 2021-09-29:
**Describe the bug**

After pulling the latest Git version via docker (1.8.2?,) I now have the detective hours icon (top right) and the User Icon->Account link available, however trying to access the URLs http://<server>:<port>/ent/subscription/#detective-hour-balance and http://<server>:<port>/ent/subscription/ returns a 404 error. Everything else is correct and works, however I don't remember seeing these options previously?

**To Reproduce**
Steps to reproduce the behavior:
1. Click Printing Hours Icon or
2. Click Username Icon -> Select account
3. ...

**Screenshots**
[image]
[image]

**Hosting environment (please complete the following information):**
 - OS: Linux Mint
 - Version [e.g. 22] 20
 - Hardware [e.g., PC, a Jetson Nano] VMWare ESXi VM

kennethjiang, 2021-10-05:
Sorry for taking so long to jump on this issue. Will fix it soon.

TempestWales, 2021-10-14:
> Please check it again @TempestWales

All sorted!