TheSpaghettiDetective/obico-server #529: [BUG] No module named 'ipware'

steros76, 2021-11-13:
I done "git pull" today and now web container explode with error "No module named 'ipware'".

I have a Debian 10 up-to-date on Intel I7.

kennethjiang, 2021-11-15:
This sounds like an environment issue. Can you ask this question in our discord server first? You will get help a lot faster that way. https://discord.gg/NcZkQfj

steros76, 2021-11-18:
I found the problem. When I deleted previous version of TSD, some hidden volumes has not been deleted.
I forced the deletion of all this volumes and I reinstalled a clear new version and now it works!

closed 2021-11-18 by steros76

kennethjiang, 2021-11-18:
Thank you for the info.

For the benefits of other users who may run into the same issue, can you share the name of the volume that you force-deleted?

steros76, 2021-11-20:
Oops. I'm sorry. I said volumes instead of images...
In docker I have only TSD and Portainer. I removed all intermediate images (docker images -a) not related to Portainer.