TheSpaghettiDetective/obico-server #535: [BUG] Share feature not working

Cralost, 2021-11-25:
**Describe the bug**
After opening shared link in browser there is an error saying "Not Found"

**To Reproduce**
Steps to reproduce the behavior:
1. open TSD 
2. click tree dots
3. Share
4. copy link
5. paste to browser

**Screenshots**
If applicable, add screenshots to help explain your problem.

App version: v1.48.75:b119

kennethjiang, 2021-11-28:
@Cralost Sorry for the delay. Can you try to share the feed in the web app to see if it works?

You can get help a lot more quickly by going to our discord server: https://discord.gg/hsMwGpD

Cralost, 2021-11-28:
So I tried and I noticed that this supposes to be premium feature - I have a free plan. So maybe that is the reason (web interface stops me).

Anyway, there could be such a pop-up as well. 

Thank  you.

kennethjiang, 2021-11-29:
Ah good point. It is a bug we overlooked. Logged it and will get it fixed in the next release.

closed 2022-04-26 by raymondh2