TheSpaghettiDetective/obico-server #540: Creating thespaghettidetective_ml_api_1 ... error

timur-tabi, 2021-11-29:
I'm following the instructions on https://www.thespaghettidetective.com/blog/2021/06/06/nvidia-jetson-nano-fun-project-3d-printing/ on my 4GB Nano (Model P3450).  I installed JetPack 4.6.  When I run install_on_jetson.sh, everything seems okay for a while, but at the end of the log I see this:

```
build this image you must use `docker-compose build` or `docker-compose up --build`.
Creating thespaghettidetective_redis_1  ... done
Creating thespaghettidetective_ml_api_1 ... error
Creating thespaghettidetective_tasks_1  ... 

ERROR: for thespaghettidetective_ml_api_1  Cannot start service ml_api: failed to create shim: OCI runtime create failed: container_linux.go:380: starting container process caused: error adding seccomp filter rule for syscall clone3: permisCreating thespaghettidetective_tasks_1  ... done

ERROR: for ml_api  Cannot start service ml_api: failed to create shim: OCI runtime create failed: container_linux.go:380: starting container process caused: error adding seccomp filter rule for syscall clone3: permission denied: unknown
ERROR: Encountered errors while bringing up the project.
```

I've attached the full build log.  I have no idea what any of this means, since I have no experience with docker.

[sd-log.txt](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/files/7615594/sd-log.txt)

kennethjiang, 2021-11-30:
Can you re-flash with JetPack 4.5.1 and try it again? Cuda is notorious for its backward compatibility (or the lack of it).

timur-tabi, 2021-11-30:
Ok, that worked, thanks.

While I have your attention, after rebooting my Nano, it doesn't appear that TSD is coming up.  I can ssh into it, but http://192.168.1.41:3334/ doesn't work.

Do I need to run some command after rebooting for start TSD?  If so, is there any way to make that automatic?

stephenjamieson, 2021-12-02:
https://github.com/NVIDIA/libnvidia-container/issues/148 - FYI I believe this is the issue, my jetson stopped working too and I automatically upgrade all packages.  I'm on 4.5.1

kennethjiang, 2021-12-10:
@stephenjamieson Not sure why I missed your message. Can you open a PR so that other users can also benefit from it?

closed 2021-12-13 by kennethjiang

kennethjiang, 2021-12-14:
@stephenjamieson Thank you for your tip at fixing this issue!

stephenjamieson, 2022-04-05:
Hey wow, sorry I missed this! I had a lot of unread emails....

commit dfffab61 (2021-12-13): Closes #540. Make it compatible with JetPack 4.6
commit 726a7ff7 (2021-12-13): Closes #540. Make it compatible with JetPack 4.6