TheSpaghettiDetective/obico-server #546: [BUG] Cannot install channels==2.4.0 and daphne 3.0.2

mrboogiee, 2021-12-13:
**Note**

 
**Describe the bug**

When executing `docker-compose up -d` the build starts and when building web the following error is received:

``` 
=> ERROR [5/6] RUN pip install -r requirements.txt
```

With details as follows:

```
#9 8.449 ERROR: Cannot install channels==2.4.0 and daphne 3.0.2 (from git+https://github.com/TheSpaghettiDetective/daphne@tsd_stable-2.5.0) because these package versions have conflicting dependencies.
#9 8.449
#9 8.449 The conflict is caused by:
#9 8.449     The user requested daphne 3.0.2 (from git+https://github.com/TheSpaghettiDetective/daphne@tsd_stable-2.5.0)
#9 8.449     channels 2.4.0 depends on daphne~=2.3
```

**To Reproduce**
Steps to reproduce the behavior:
1. clone or pull latest version (7ab6a8d5c3ba3064e0e5308b26dbc65098eeda62) on amd64 system
2. execute `docker-compose up -d`
3. watch the build of web fail

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hosting environment (please complete the following information):**
 - OS: [e.g. iOS]
 tested on both Synology OS and MacOS (Intel)
 - Version [e.g. 22]
 latest
 - Hardware [e.g., PC, a Jetson Nano]
 NAS and laptop

**Additional context**
Add any other context about the problem here.
full log details:
```
Creating network "thespaghettidetective_default" with the default driver
Building web
[+] Building 15.7s (9/10)
 => [internal] load build definition from Dockerfile                                                                                                                                                                                                                                                                                                                                                                                                                                              0.2s
 => => transferring dockerfile: 37B                                                                                                                                                                                                                                                                                                                                                                                                                                                               0.1s
 => [internal] load .dockerignore                                                                                                                                                                                                                                                                                                                                                                                                                                                                 0.1s
 => => transferring context: 34B                                                                                                                                                                                                                                                                                                                                                                                                                                                                  0.1s
 => [internal] load metadata for docker.io/thespaghettidetective/web:base-1.7                                                                                                                                                                                                                                                                                                                                                                                                                     0.6s
 => [internal] load build context                                                                                                                                                                                                                                                                                                                                                                                                                                                                 6.2s
 => => transferring context: 36.18kB                                                                                                                                                                                                                                                                                                                                                                                                                                                              6.2s
 => [1/6] FROM docker.io/thespaghettidetective/web:base-1.7@sha256:1df3ce4409ae9296909cc91a8594720b484d21210de44d10913e8bbc9e60e99d                                                                                                                                                                                                                                                                                                                                                               0.0s
 => CACHED [2/6] WORKDIR /app                                                                                                                                                                                                                                                                                                                                                                                                                                                                     0.0s
 => CACHED [3/6] ADD . /app                                                                                                                                                                                                                                                                                                                                                                                                                                                                       0.0s
 => CACHED [4/6] RUN pip install -U pip                                                                                                                                                                                                                                                                                                                                                                                                                                                           0.0s
 => ERROR [5/6] RUN pip install -r requirements.txt                                                                                                                                                                                                                                                                                                                                                                                                                                               8.7s
------
 > [5/6] RUN pip install -r requirements.txt:
#9 0.769 Collecting git+https://github.com/TheSpaghettiDetective/daphne@tsd_stable-2.5.0 (from -r requirements.txt (line 29))
#9 0.770   Cloning https://github.com/TheSpaghettiDetective/daphne (to revision tsd_stable-2.5.0) to /tmp/pip-req-build-0pgrbc7j
#9 0.774   Running command git clone --filter=blob:none -q https://github.com/TheSpaghettiDetective/daphne /tmp/pip-req-build-0pgrbc7j
#9 1.898   Running command git checkout -b tsd_stable-2.5.0 --track origin/tsd_stable-2.5.0
#9 2.291   Branch 'tsd_stable-2.5.0' set up to track remote branch 'tsd_stable-2.5.0' from 'origin'.
#9 2.291   Switched to a new branch 'tsd_stable-2.5.0'
#9 2.291   Resolved https://github.com/TheSpaghettiDetective/daphne to commit 9b4679beb88f137162b7f5f21eccfabd2b7e8805
#9 2.293   Preparing metadata (setup.py): started
#9 3.473   Preparing metadata (setup.py): finished with status 'done'
#9 3.476 Collecting git+https://github.com/ivahotin/django-channels-presence.git (from -r requirements.txt (line 38))
#9 3.477   Cloning https://github.com/ivahotin/django-channels-presence.git to /tmp/pip-req-build-ybo6tfyj
[... 132 lines cut ...]
#9 8.002   Using cached dj_database_url-0.5.0-py2.py3-none-any.whl (5.5 kB)
[... cut ...]
#9 8.449     channels 2.4.0 depends on daphne~=2.3
#9 8.449
#9 8.449 To fix this you could try to:
#9 8.449 1. loosen the range of package versions you've specified
#9 8.449 2. remove package versions to allow pip attempt to solve the dependency conflict
#9 8.449
#9 8.449 ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/user_guide/#fixing-conflicting-dependencies
------
executor failed running [/bin/sh -c pip install -r requirements.txt]: exit code: 1
ERROR: Service 'web' failed to build : Build failed
```

mrboogiee, 2021-12-13:
just tried, commit 54b368a5304ef1a6cf17e9a068545f7f81e41ed4 doesn't have this issue.

encetamasb, 2021-12-13:
@mrboggie Hi! Thanks for reporting this issue, I've found the problem in our daphne fork. It should work now.

kennethjiang, 2021-12-14:
@mrboogiee Closing the issue as it should be fixed now.