TheSpaghettiDetective/obico-server #548: [BUG] Docker fails to start on Jetson Nano

Sudrien, 2021-12-17:
https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/docs/jetson_guide.md and
https://www.thespaghettidetective.com/blog/2021/06/06/nvidia-jetson-nano-fun-project-3d-printing/

have pretty much the same setup

> ERROR: Invalid interpolation format for "build" option in service "web": "Dockerfile${DOCKERFILE_VERSION-}"

Not being familiar with Docker, it's "DOCKERFILE_VERSION" in the release tag, and the only _other_ place _that_ shows up is the Makefile.

When I try that tag - release - 

WARNING: The DOCKERFILE_VERSION variable is not set. Defaulting to a blank string.
ERROR: Version in "./docker-compose.yml" is unsupported. You might be seeing this error because you're using the wrong Compose file version. Either specify a supported version (e.g "2.2" or "3.3") and place your service definitions under the `services` key, or omit the `version` key and place your service definitions at the root of the file to use version 1.
For more on the Compose file format versions, see https://docs.docker.com/compose/compose-file/

Sudrien, 2021-12-17:
:~/TheSpaghettiDetective$ dpkg-query --show nvidia-l4t-core
nvidia-l4t-core	32.6.1-20210726122000

I reading this mean Jetpack 4.6

kennethjiang, 2021-12-20:
This looks like an issue with your setup. The quickest way to get help with issues like this is to ask our discord community: https://discord.gg/hsMwGpD

Sudrien, 2022-01-24:
Findings: I am limited to pre-Jetpack 4.6

Which means, use an older image when flashing; 

https://developer.download.nvidia.com/embedded/L4T/r32_Release_v5.1/r32_Release_v5.1/Jeston_Nano/jetson-nano-jp451-sd-card-image.zip

THEN, once booted,

Go into `software updater` before running security updates, and disabling nvidia sources. Otherwise it will try and pull newer jetpack components, which it will fail to configure.

closed 2022-01-24 by Sudrien

Sudrien, 2022-01-27:
An extra note: 

If running TSD & Octoprint on the same Nano, make sure to use not-127.0.0.1-IP and not hostname when pointing them at each other. Docker needs to be configured to resolve to the host, ip, and currently it is not.

kennethjiang, 2022-01-29:
@Sudrien Great you figured this out! And thank you so much for sharing the info with the TSD community!