TheSpaghettiDetective/obico-server #551: Filter for 'focused review needed' not functional

marcelrv, 2021-12-22:
**Describe the bug**
Filter for 'focused review needed' shows also items without the focused review needed in the screen.
 
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. load the regular screen
2. select filter in the top right of the screen
3. expect the screen to show only the tumbs according to the filter

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hosting environment (please complete the following information):**
 - OS: Linux/docker image. Self hosted version
 - Version latest from github (22-12-2021)
 - Hardware  PC, 

**Additional context**
Add any other context about the problem here.

kennethjiang, 2021-12-23:
Can you click on the thumb up/down buttons on these timelapses and see if the focused feedback button will show up?

marcelrv, 2021-12-23:
Yes, it shows. There are a few in the list, but I could not get them in the screenshot.

Clicking on them indeed shows the dialog, you click through the various pictures and have them evaluated and submitted at the end. As far as I can see similar to the tsd hosted version.

kennethjiang, 2022-01-01:
@marcelrv Sorry just got back from a vacation.

The focused feedback option won't show unless the thumb up/down is pressed. This is true for both the private server and the TSD cloud.

kennethjiang, 2022-01-01:
Since this is the expected behavior, I'm closing the issue.

closed 2022-01-01 by kennethjiang