TheSpaghettiDetective/obico-server #552: [BUG] app hangs on fresh install of recent Jetson Nano images

smartin015, 2022-01-03:
Summary of discussion from discord where I and another user (TargetCrafter) are both encountering this problem.
 
**Describe the bug**
Running TSD on my jetson nano (4GB) from a fresh installation works for a bit (~hours to a day or two), but eventually it goes unresponsive: octoprint can't contact the server, and I can't access the webpage (:3334). 

Completely removing and reinstalling the repo and re-linking the octoprint plugin gets it working again, I can also sometimes get it to recover with a full down/up via docker-compose.

The issue appears to be a mismatch in CUDA versions between the cuda driver running the model vs the installed runtime, per logs in the `ml_api` container:
```
/usr/bin/python3: [2022-01-02 16:17:44 +0000] [4128] [INFO] Booting worker with pid: 4128
layer     filters    size              input                output
 Try to load cfg: /app/model/model.cfg, weights: /app/model/model.weights, clear = 0
CUDA status Error: file: ./src/cuda.c : () : line: 203 : build time: Oct 15 2019 - 16:00:39
CUDA Error: CUDA driver version is insufficient for CUDA runtime version
   0  Try to set subdivisions=64 in your cfg-file.
```

Replacing usage of the jetsistant docker image in the ml_api base image with the official L4T base image (nvcr.io/nvidia/l4t-base:r32.4.3) produces an error indicating the model expects CUDART v10: https://pastebin.com/z5QcD48S. Attempting to use the repo fork of jetsistant (https://github.com/clearobject/docker-cuda-jetpack) yields the same result. 

I suspect the model must be rebuilt to work with the new version of cuda. I have no idea how to do this, otherwise I would volunteer a PR.

**To Reproduce**
Steps to reproduce the behavior:
1. Install the [newest image of the jetson nano developer kit](https://developer.nvidia.com/embedded/learn/get-started-jetson-nano-devkit#write)
2. Install and configure TSD as per [the guide](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/docs/jetson_guide.md)
3. Wait a few hours

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hosting environment (please complete the following information):**
 - OS: Jetson Nano 4gb
 - Version: JetPack 4.6
 - Hardware: Jetson nano

```shell
jetson1@jetson1:~$ dpkg-query --show nvidia-l4t-core
nvidia-l4t-core    32.6.1-20210726122000
```

(Corresponding to JetPack 4.6 - seems to be the source of the "driver version" errors in the ML API per https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/issues/515 but they don't seem to prevent it working initially)

**Additional context**

web container logs: https://pastebin.com/QZpV76nu
ML API logs: https://pastebin.com/yhQea6iA
Tasks container logs: https://pastebin.com/XNnpYEMq
Redis logs: https://pastebin.com/DHTPzasb

smartin015, 2022-01-06:
Hey @kennethjiang - I have a little prior ML experience and would be happy to try my hand at adding compatibility for newer versions of JetPack, but I need some help understanding how the model is currently built (specifically, how to build the `ml_api/bin/model_*.so` binaries as they seem to be the source of the dependency on CUDART 10).

It looks like the only related files in this repo are for loading and running the built model, not for building/training it. Do you have any pointers?

kennethjiang, 2022-01-08:
@smartin015 To be honest I don't remember exactly how I built the binary. But I'm pretty sure I just follow the instruction at: https://github.com/AlexeyAB/darknet. Let me know if you have any luck with it!

smartin015, 2022-01-10:
Okay, hacked it up and running again on my install of Jetpack 4.6. I'm not 100% happy with how my fix is handling `*.so` dependencies of the darknet binary, but it does indeed run with no errors. 

The changes: https://github.com/smartin015/TheSpaghettiDetective/commit/e41e6bf21feb95e51784aa40a6f9afdf066dd7a9

Notes on what I did: https://github.com/smartin015/TheSpaghettiDetective/blob/e41e6bf21feb95e51784aa40a6f9afdf066dd7a9/docs/jetson_guide.md#troubleshooting--advanced-model-details

@kennethjiang do you know of any reasons why libcuda.so.1 and libnvidia-ptxjitcompiler.so.1 would exist in `/usr/local/aarch64-linux-gnu/tegera` on the jetson itself, but not when I run e.g. `nvidia-docker run -it --rm cudajetpack:4.5runtime /bin/bash`, despite the more specific versions still existing (i.e. `libnvidia-ptxjitcompiler.so.440.18` and `libcuda.so.1.1`)? I have to link them manually every time I start the container. If we can resolve that, I'd be happy to contribute a proper pull request for other folks to use.

kennethjiang, 2022-01-12:
@smartin015 It looks like you have the building blocks for a good PR.

I don't know why you have to link these libs everytime. It feels to me just yet another quirkiness about cuda. One way to work around it is to build these links in the image. Basically put these commands in the `Dockerfile.base_aarch64` instead of the `docker-compose.yaml` file.

Also please test if your changes are compatible with JetPack 4.5. If not, we will need to change that part of the help doc.

jonwilliams84, 2022-01-12:
@smartin015 Thanks for the pointer...JP4.6 working for me now. 

However its worth point out I had missing links for a few more libraries.

To resolve I added the following to the `mlapi/Dockerfile` (I too didn't like the docker-compose command hack :-)):

```
RUN ln -sf /usr/lib/aarch64-linux-gnu/tegra/libcuda.so /usr/lib/aarch64-linux-gnu/tegra/libcuda.so.1 && \
    ln -s /usr/lib/aarch64-linux-gnu/tegra/libnvidia-ptxjitcompiler.so.440.18 /usr/lib/aarch64-linux-gnu/tegra/libnvidia-ptxjitcompiler.so.1 && \
    ln -s /usr/local/cuda-10.2/targets/aarch64-linux/lib/libcudart.so.10.2 /usr/lib/aarch64-linux-gnu/tegra/libcudart.so.10.2 && \
    ln -s /usr/local/cuda-10.2/targets/aarch64-linux/lib/libcublas.so.10 /usr/lib/aarch64-linux-gnu/tegra/libcublas.so.10 && \
    ln -s /usr/local/cuda-10.2/targets/aarch64-linux/lib/libcurand.so.10 /usr/lib/aarch64-linux-gnu/tegra/libcurand.so.10 && \
    ln -s /usr/local/cuda-10.2/targets/aarch64-linux/lib/libcublasLt.so.10 /usr/lib/aarch64-linux-gnu/tegra/libcublasLt.so.10
```
Then rebuilt with `docker-compose build ml_api`

smartin015, 2022-01-15:
Nice fix! Though I'm a bit surprised linking at build time worked successfully... I thought `nvidia-docker` was being sneaky with container setup and adding in OS-specific files at runtime. 

I'll attempt to modify my setup and send a PR if it works :)

smartin015, 2022-01-15:
@jonwilliams84, I get 

`ln: failed to create symbolic link '/usr/lib/aarch64-linux-gnu/tegra/libcuda.so.1': No such file or directory`

When I try `docker-compose build ml_api`. Can you share:
* your `Dockerfile.base_aarch64` and `ml_api/Dockerfile`
* the exact build command you ran to build the base image
* Results of `lsb_release -a`, `uname -a`, and `docker --version`

There must be some sort of difference in our setups...

jonwilliams84, 2022-01-31:
Actually @smartin015  - Looking back through my notes I build with: `docker-compose build ml_api` from the base TSD directory.

Either should work though

advisoray, 2022-02-08:
I'm having the same issue as @smartin015: `ln: failed to create symbolic link '/usr/lib/aarch64-linux-gnu/tegra/libcuda.so.1': No such file or directory`.  I'm also on jetpack 4.6 and seeing the same results.

`docker version`
```
Client:
 Version:           20.10.7
 API version:       1.41
 Go version:        go1.13.8
 Git commit:        20.10.7-0ubuntu5~18.04.3
 Built:             Mon Nov  1 01:04:31 2021
 OS/Arch:           linux/arm64
 Context:           default
 Experimental:      true

Server:
 Engine:
  Version:          20.10.7
  API version:      1.41 (minimum version 1.12)
  Go version:       go1.13.8
  Git commit:       20.10.7-0ubuntu5~18.04.3
  Built:            Fri Oct 22 00:57:37 2021
  OS/Arch:          linux/arm64
  Experimental:     false
 containerd:
  Version:          1.5.5-0ubuntu3~18.04.1
  GitCommit:        
 runc:
  Version:          1.0.1-0ubuntu2~18.04.1
  GitCommit:        
 docker-init:
  Version:          0.19.0
  GitCommit:  
```

`docker --version`:
```
Docker version 20.10.7, build 20.10.7-0ubuntu5~18.04.3
```

`uname -a`:
```
Linux jetsonnano 4.9.253-tegra #1 SMP PREEMPT Mon Jul 26 12:13:06 PDT 2021 aarch64 aarch64 aarch64 GNU/Linux
```

`lsb_release -a`:
```
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.6 LTS
Release:	18.04
Codename:	bionic
```

jonwilliams84, 2022-02-09:
Weird.  Just rebuilt and mine worked.

Maybe try a find and update the path:
```
find / -name libcuda.so 2>/dev/null
/usr/local/cuda-10.2/targets/aarch64-linux/lib/stubs/libcuda.so
/usr/lib/aarch64-linux-gnu/tegra/libcuda.so
/usr/lib/aarch64-linux-gnu/libcuda.so
```

advisoray, 2022-02-09:
Very strange. Running the `find` command returns the same results as you.
```
find / -name libcuda.so 2>/dev/null
/usr/local/cuda-10.2/targets/aarch64-linux/lib/stubs/libcuda.so
/usr/lib/aarch64-linux-gnu/tegra/libcuda.so
/usr/lib/aarch64-linux-gnu/libcuda.so
```

jonwilliams84, 2022-02-10:
@advisoray  - perms maybe? Don't see logical explanation as to why but I have seen stranger things in the past.

smartin015, 2022-02-11:
Mine returns in reverse order... /shrug
```
find / -name libcuda.so 2>/dev/null
/usr/lib/aarch64-linux-gnu/libcuda.so
/usr/lib/aarch64-linux-gnu/tegra/libcuda.so
/usr/local/cuda-10.2/targets/aarch64-linux/lib/stubs/libcuda.so
```

But it's the same files. That may all be irrelevant though if the build image doesn't link them in. Adding `RUN ls -al /usr/local/aarch64-linux-gnu/tegra/` to the `Dockerfile` yields:

```
Step 8/9 : RUN ls -al /usr/lib/aarch64-linux-gnu/tegra/
 ---> Running in e5bd2709f529
ls: cannot access '/usr/lib/aarch64-linux-gnu/tegra/': No such file or directory
```

FWIW my `docker --version` and `docker version` exactly match that of @jonwilliams84: (`Docker version 20.10.7, build 20.10.7-0ubuntu5~18.04.3` etc etc)

@advisoray - I copied and pasted your Dockerfile directly - still get an error of missing files. 

Trying with a full revert of `Dockerfile.base_aarch64` (back to using the jetsistant image) yields the same result.

If it's this flaky across 3 similar setups, I think we shouldn't trust linking at build time to work as a general fix. But we can do something similar - since ml_api is invoked by `bash -c`, it runs `~/.bashrc` on container startup. IMO it's not so different to just `RUN echo "ln -s ...." >> ~/.bashrc` in the Dockerfile, which moves the linking to be done at container startup and works just fine on my setup.

However - I'm beginning to suspect there's something else going on here. I've had a few prints failing with spaghetti that the detective hasn't caught. I'm now digging into `server.py` and `detection_model.py` and will report back what I find.

smartin015, 2022-02-12:
Turns out, there were some extensions made to the [`detection` ctype](https://github.com/AlexeyAB/darknet/blob/master/darknet.py#L24) which caused improper access in the original python code (compare vs [here](https://github.com/pjreddie/darknet/blob/a3714d0a2bf92c3dea84c4bea65b2b0c64dbc6b1/python/darknet.py#L26)).

Copying in the type changes fixes some NULL pointer access issues I was seeing once I fixed the logging, and I do indeed see detections but the probabilities are all zero so I don't think it's quite fixed yet. 

There's also a bunch of other improvements made in the AlexeyAB fork, including a module which could basically replace `lib/detection_model.py`, but I'm hesitant to change too much at once and risk breaking compatibility elsewhere.

smartin015, 2022-02-18:
Brief update - I'm still looking into why there aren't any detections. To speed things along, I'm dusting off the testing script in `ml_api/scripts/detect_timelapse.sh`, but it appears to have bitrotted a bit (undocumented, wrong paths, missing deps vs what's installed in the container etc). I've also tried the latest python scripts from AlexeyAB/darknet, but I get segfaults when I run it. 

More digging to do :)

smartin015, 2022-02-18:
Hey @kennethjiang, I'm afraid I'm at a loss. Even running the model on my workstation with a fresh compiled version of darknet produces no detections. Can you please advise? Am I doing something wrong? What validation images do you use to confirm that spaghetti is detected?

Steps to reproduce:

```
# Clone TSD repo and get weights
git clone https://github.com/TheSpaghettiDetective/TheSpaghettiDetective && cd TheSpaghettiDetective
cd ml_api/model && wget https://tsd-pub-static.s3.amazonaws.com/ml-models/3209.neg_32213.22300.weights && cd ../../

# Clone and build darknet
git clone https://github.com/AlexeyAB/darknet
cd darknet
vim Makefile 
# Here I set GPU=1
make -j$(nproc)

# Some file repair/path tweaks
vim darknet_images.py
# For some reason darknet_images.py has `import darknet` in two places; remove the first one to prevent https://github.com/opencv/opencv/issues/14884
vim ../ml_api/model/model.meta
# Change path to "names" to be something like /home/foo/.../TheSpaghettiDetective/ml_api/model/names

./darknet detector test ../ml_api/model/model.meta ../ml_api/model/model.cfg ../ml_api/model/3209.neg_32213.22300.weights -thresh 0.1 ~/Downloads/spaghetti.png
```

The result I get:
```
 CUDA-version: 11020 (11020), GPU count: 1  
 OpenCV isn't used - data augmentation will be slow 
 0 : compute_capability = 610, cudnn_half = 0, GPU: GeForce GTX 1070 
net.optimized_memory = 0 
mini_batch = 1, batch = 8, time_steps = 1, train = 0 
   layer   filters  size/strd(dil)      input                output
   0 Create CUDA-stream - 0 
conv     32       3 x 3/ 1    416 x 416 x   3 ->  416 x 416 x  32 0.299 BF
   1 max                2x 2/ 2    416 x 416 x  32 ->  208 x 208 x  32 0.006 BF
   2 conv     64       3 x 3/ 1    208 x 208 x  32 ->  208 x 208 x  64 1.595 BF
   3 max                2x 2/ 2    208 x 208 x  64 ->  104 x 104 x  64 0.003 BF
   4 conv    128       3 x 3/ 1    104 x 104 x  64 ->  104 x 104 x 128 1.595 BF
   5 conv     64       1 x 1/ 1    104 x 104 x 128 ->  104 x 104 x  64 0.177 BF
   6 conv    128       3 x 3/ 1    104 x 104 x  64 ->  104 x 104 x 128 1.595 BF
   7 max                2x 2/ 2    104 x 104 x 128 ->   52 x  52 x 128 0.001 BF
   8 conv    256       3 x 3/ 1     52 x  52 x 128 ->   52 x  52 x 256 1.595 BF
   9 conv    128       1 x 1/ 1     52 x  52 x 256 ->   52 x  52 x 128 0.177 BF
  10 conv    256       3 x 3/ 1     52 x  52 x 128 ->   52 x  52 x 256 1.595 BF
  11 max                2x 2/ 2     52 x  52 x 256 ->   26 x  26 x 256 0.001 BF
  12 conv    512       3 x 3/ 1     26 x  26 x 256 ->   26 x  26 x 512 1.595 BF
  13 conv    256       1 x 1/ 1     26 x  26 x 512 ->   26 x  26 x 256 0.177 BF
  14 conv    512       3 x 3/ 1     26 x  26 x 256 ->   26 x  26 x 512 1.595 BF
  15 conv    256       1 x 1/ 1     26 x  26 x 512 ->   26 x  26 x 256 0.177 BF
  16 conv    512       3 x 3/ 1     26 x  26 x 256 ->   26 x  26 x 512 1.595 BF
  17 max                2x 2/ 2     26 x  26 x 512 ->   13 x  13 x 512 0.000 BF
  18 conv   1024       3 x 3/ 1     13 x  13 x 512 ->   13 x  13 x1024 1.595 BF
  19 conv    512       1 x 1/ 1     13 x  13 x1024 ->   13 x  13 x 512 0.177 BF
  20 conv   1024       3 x 3/ 1     13 x  13 x 512 ->   13 x  13 x1024 1.595 BF
  21 conv    512       1 x 1/ 1     13 x  13 x1024 ->   13 x  13 x 512 0.177 BF
  22 conv   1024       3 x 3/ 1     13 x  13 x 512 ->   13 x  13 x1024 1.595 BF
  23 conv   1024       3 x 3/ 1     13 x  13 x1024 ->   13 x  13 x1024 3.190 BF
  24 conv   1024       3 x 3/ 1     13 x  13 x1024 ->   13 x  13 x1024 3.190 BF
  25 route  16 		                           ->   26 x  26 x 512 
  26 conv     64       1 x 1/ 1     26 x  26 x 512 ->   26 x  26 x  64 0.044 BF
  27 
 reorg_old 
reorg_old              / 2    26 x  26 x  64   ->    13 x  13 x 256
  28 route  27 24 	                           ->   13 x  13 x1280 
  29 conv   1024       3 x 3/ 1     13 x  13 x1280 ->   13 x  13 x1024 3.987 BF
  30 conv     30       1 x 1/ 1     13 x  13 x1024 ->   13 x  13 x  30 0.010 BF
  31 detection
mask_scale: Using default '1.000000'
Total BFLOPS 29.338 
avg_outputs = 607364 
 Allocate additional workspace_size = 49.84 MB 
Loading weights from ../ml_api/model/3209.neg_32213.22300.weights...
 seen 64, trained: 1427 K-images (22 Kilo-batches_64) 
Done! Loaded 32 layers from weights-file 
 Detection layer: 31 - type = 27 
/home/semartin/Downloads/spaghetti.png: Predicted in 298.118000 milli-seconds.
Not compiled with OpenCV, saving to predictions.png instead
```

`predictions.png` has zero boxes drawn, and there's no indication of any detections in the console output. You can see the `spaghetti.png` image I used below:

[image]

smartin015, 2022-02-19:
Running `docker-compose exec ml_api bash` just terminates without output. Should I not be able to just load the model weights, config, etc. into darknet and run it via the `./darknet` binary?

It would also help greatly to have an image and threshold value which is known to cause the model to detect a failure, so that I can be sure I'm debugging the model and not the image.

kennethjiang, 2022-02-19:
> Running `docker-compose exec ml_api bash` on my workstation after running `docker-compose build ml_api` on an unchanged fresh pull of the master branch of TSD just terminates without output.

It seems that your TSD cluster has some issues. You will eventually need to debug that. Maybe you can recreate the cluster on a PC so that you will at least have a working cluster to compare to.

> 
> It would also help greatly to have an image and threshold value which is known to cause the model to detect a failure, so that I can be sure I'm debugging the model and not the image.

Unfortunately there are no such "magic" thresholds. The detection itself is adaptive. The relevant code can be found at: https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/web/lib/prediction.py#L21

> The model still isn't detecting anything in the image. Do you have a golden "failure" image I can detect against?

There are no golden images either. But you can download one of the videos at https://app.thespaghettidetective.com/ent_pub/publictimelapses/ and extract the images yourself.

smartin015, 2022-02-20:
> It seems that your TSD cluster has some issues. You will eventually need to debug that. Maybe you can recreate the cluster on a PC so that you will at least have a working cluster to compare to.

This was on a PC - on my workstation, running Ubuntu 21.04 with a GTX 1070 GPU.

> Unfortunately there are no such "magic" thresholds. The detection itself is adaptive. The relevant code can be found at: [...]

I understand that you're adjusting the threshold of the model dynamically. What I want is **any indication at all** that the model can detect spaghetti. 

> > The model still isn't detecting anything in the image. Do you have a golden "failure" image I can detect against?
> 
> There are no golden images either. 

The model should be deterministic. It's just a CNN with no memory. If I had an image of spaghetti, and a threshold value (such as 0.02 - very low certainty!) I should expect it to detect *something* every time when I ran on it. In my debugging I'm looking for any output at all, to confirm that I've indeed loaded a model that can detect spaghetti.

> But you can download one of the videos at https://app.thespaghettidetective.com/ent_pub/publictimelapses/ and extract the images yourself.

Those videos include detection overlays. I wasn't certain whether that would foul up detection or not, but trying anyway with this one:

[image]

I do get a result at least. For posterity, this is what I ran (**on my PC**):

```
sudo docker run -it --rm --gpus=all thespaghettidetective_ml_api:latest /bin/bash
wget https://user-images.githubusercontent.com/607666/154857506-f67fe00c-a423-4a12-9ee0-8dced1b72968.png
python -m lib.detection_model 154857506-f67fe00c-a423-4a12-9ee0-8dced1b72968.png
```

And the output:
```
Try to load cfg: model/model.cfg, weights: model/model.weights, clear = 0 
layer     filters    size              input                output
   0 conv     32  3 x 3 / 1   416 x 416 x   3   ->   416 x 416 x  32 0.299 BF
   1 max          2 x 2 / 2   416 x 416 x  32   ->   208 x 208 x  32 0.006 BF
   2 conv     64  3 x 3 / 1   208 x 208 x  32   ->   208 x 208 x  64 1.595 BF
   3 max          2 x 2 / 2   208 x 208 x  64   ->   104 x 104 x  64 0.003 BF
   4 conv    128  3 x 3 / 1   104 x 104 x  64   ->   104 x 104 x 128 1.595 BF
   5 conv     64  1 x 1 / 1   104 x 104 x 128   ->   104 x 104 x  64 0.177 BF
   6 conv    128  3 x 3 / 1   104 x 104 x  64   ->   104 x 104 x 128 1.595 BF
   7 max          2 x 2 / 2   104 x 104 x 128   ->    52 x  52 x 128 0.001 BF
   8 conv    256  3 x 3 / 1    52 x  52 x 128   ->    52 x  52 x 256 1.595 BF
   9 conv    128  1 x 1 / 1    52 x  52 x 256   ->    52 x  52 x 128 0.177 BF
  10 conv    256  3 x 3 / 1    52 x  52 x 128   ->    52 x  52 x 256 1.595 BF
  11 max          2 x 2 / 2    52 x  52 x 256   ->    26 x  26 x 256 0.001 BF
  12 conv    512  3 x 3 / 1    26 x  26 x 256   ->    26 x  26 x 512 1.595 BF
  13 conv    256  1 x 1 / 1    26 x  26 x 512   ->    26 x  26 x 256 0.177 BF
  14 conv    512  3 x 3 / 1    26 x  26 x 256   ->    26 x  26 x 512 1.595 BF
  15 conv    256  1 x 1 / 1    26 x  26 x 512   ->    26 x  26 x 256 0.177 BF
  16 conv    512  3 x 3 / 1    26 x  26 x 256   ->    26 x  26 x 512 1.595 BF
  17 max          2 x 2 / 2    26 x  26 x 512   ->    13 x  13 x 512 0.000 BF
  18 conv   1024  3 x 3 / 1    13 x  13 x 512   ->    13 x  13 x1024 1.595 BF
  19 conv    512  1 x 1 / 1    13 x  13 x1024   ->    13 x  13 x 512 0.177 BF
  20 conv   1024  3 x 3 / 1    13 x  13 x 512   ->    13 x  13 x1024 1.595 BF
  21 conv    512  1 x 1 / 1    13 x  13 x1024   ->    13 x  13 x 512 0.177 BF
  22 conv   1024  3 x 3 / 1    13 x  13 x 512   ->    13 x  13 x1024 1.595 BF
  23 conv   1024  3 x 3 / 1    13 x  13 x1024   ->    13 x  13 x1024 3.190 BF
  24 conv   1024  3 x 3 / 1    13 x  13 x1024   ->    13 x  13 x1024 3.190 BF
  25 route  16
  26 conv     64  1 x 1 / 1    26 x  26 x 512   ->    26 x  26 x  64 0.044 BF
  27 reorg              / 2    26 x  26 x  64   ->    13 x  13 x 256
  28 route  27 24
  29 conv   1024  3 x 3 / 1    13 x  13 x1280   ->    13 x  13 x1024 3.987 BF
  30 conv     30  1 x 1 / 1    13 x  13 x1024   ->    13 x  13 x  30 0.010 BF
  31 detection
mask_scale: Using default '1.000000'
Total BFLOPS 29.338 
Loading weights from model/model.weights...
 seen 64 
Done!
Loaded - names_list: /app/model/names, classes = 1 
[('failure', 0.30526742339134216, (1877.6353759765625, 787.1133422851562, 130.99317932128906, 192.90467834472656))]
```

I can now also confirm detections running exclusively with darknet on the jetson nano (as in my [previous comment](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/issues/552#issuecomment-1044979606)). 

Now that I have a testable setup, I can simplify down and get something that works without a substantial refactor to the latest darknet code.

kennethjiang, 2022-02-21:
Good to know you are making progress.

> As a side note, it would be excellent if this repo came with one or more unit tests that confirmed detection - whether via video, or just via still image as above.

Unit tests are usually created for code issues, not issues with a dev environment. But I might have misunderstood what you mean to say here. Maybe you can come up with something and open a PR?

smartin015, 2022-02-25:
I believe I've fixed the model issues in [my jetpack_fix branch](https://github.com/smartin015/TheSpaghettiDetective/tree/jetpack_fix). Much cleanup needed before I contribute a PR - and I'd like to also run it for a bit and ensure it can actually catch some real cases of spaghetti on my printer.

@kennethjiang, a couple questions for you:

1. At the heart of my fix is a transition to reusing [darknet_images.py](https://github.com/AlexeyAB/darknet/blob/master/darknet_images.py) from AlexeyAB's darknet repo. I'm sure it's possible to fix up the detection model code and incur a bit less change, but that seems like unnecessary parallel development if darknet can already do it for you. **WDYT about using `darknet_images.py` directly from the darknet repo** and eliminating [lib/detection_model.py](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/ml_api/lib/detection_model.py)? This would also affect non-Jetson devices and would need to be tested on whatever systems y'all use to run it as a service.
2. I've been putting lessons learned at the bottom of [jetson_guide.md](https://github.com/smartin015/TheSpaghettiDetective/blob/a5d890c51a01e85c6b920bbad937ed0c36f3eb92/docs/jetson_guide.md), but a lot of it is general advice for developing on the model. **I recommend adding a `docs/model_development.md` file** and putting the cleaned up instructions there. Is this OK? This can be an independent PR from the rest of the changes.
3. I made some changes to `server.py` to allow for more detailed logging - it generates ~3 log outputs per request. **Should the more verbose ml_api logging be enabled or disabled by default?**

Re: unit tests, perhaps "e2e tests" might be a better phrasing. It'd be useful to know after a code change or model training run that, for a set of "golden" test pictures:

1. The model detects spaghetti when there is spaghetti in a picture
2. The model does not detect spaghetti when there is not spaghetti in a picture

Which would also shake out bugs where the model or shared libraries fail to load, interfaces/object structures change etc.

Thanks!

kennethjiang, 2022-02-26:
> I believe I've fixed the model issues in [my jetpack_fix branch](https://github.com/smartin015/TheSpaghettiDetective/tree/jetpack_fix). Much cleanup needed before I contribute a PR - and I'd like to also run it for a bit and ensure it can actually catch some real cases of spaghetti on my printer.
> 
Great.

> 1. At the heart of my fix is a transition to reusing [darknet_images.py](https://github.com/AlexeyAB/darknet/blob/master/darknet_images.py) from AlexeyAB's darknet repo. I'm sure it's possible to [...]
This sounds good. `darknet_images.py` didn't exist previously so I had to create our own version.

> 2. I've been putting lessons learned at the bottom of [jetson_guide.md](https://github.com/smartin015/TheSpaghettiDetective/blob/a5d890c51a01e85c6b920bbad937ed0c36f3eb92/docs/jetson_guide.md), but a [...]
Yes please separate them into a different guide. Let's keep `jetson_guide.md` for things specific to Jetson.

> 3. I made some changes to `server.py` to allow for more detailed logging - it generates ~3 log outputs per request. **Should the more verbose ml_api logging be enabled or disabled by default?**
They should be disabled by default.

> Re: unit tests, perhaps "e2e tests" might be a better phrasing. It'd be useful to know after a code change or model training run that, for a set of "golden" test pictures:
>
> 1. The model detects [...]
Feel free to create the tests and send them in as a separate PR.

smartin015, 2022-03-02:
Update: Opened https://github.com/AlexeyAB/darknet/pull/8387 to upstream a change in the darknet repo - will give it a week or so to see if they'll accept it - in the meantime I'll likely copy out the `image_detection` function with some slight tweaks. We can remove the slight duplication if/when that PR is merged.

After running things for a bit, I've confirmed that print failures are detected and the printer is appropriately paused. 

My immediate next step will be submitting a new guide for exploring/developing the model.

kennethjiang, 2022-03-04:
@smartin015 Awesome! Sounds good to duplicate the code for now and removing the duplicates once your PR is accepted.

smartin015, 2022-04-06:
Friendly update: no progress of late as I've been occupied in other projects. Currently getting my attention back on #589 - when that's reached a good state I'll loop back around here and work on the `darknet_images` replacement.

advisoray, 2022-05-15:
Is there anything I can do to help out on this?

kennethjiang, 2022-05-16:
@advisoray Thank you for offering to help. Feel free to take a stab at it. You can take a look [at the work-in-progress](https://github.com/smartin015/TheSpaghettiDetective/tree/jetpack_fix) to see if it'll give you a jump-start. We really need help from the community for this one as handling Jetson Nano is very time-consuming and we simply don't have enough bandwidth for now.

smartin015, 2022-05-16:
What Kenneth said - see also comments on #614. I think the last sticking point was in arranging the dockerfiles and maintaining CPU support. Been meaning to get to this, but it'll likely be a few more weeks before some free time opens up, so please take a crack at it :)

smartin015, 2022-05-17:
Also note that https://github.com/AlexeyAB/darknet/pull/8387 just merged into Darknet, so you should be able to remove `image_detection` defined in #614 and replace it with `darknet.image_detection()` for even less code to maintain :)

smartin015, 2022-06-06:
Hey @advisoray, did you make any progress on this? I have a bit more free time to work on this, but I don't want to stomp on any progress you've made.

advisoray, 2022-06-06:
Nope, no progress so far. I've been in the middle of a move and have had
less free time than I wanted to work on it.

smartin015, 2022-09-02:
Just leaving a note here - #614 was ready to go and worked for Jetpack 4.6, but stalled when potential compatibility problems were encountered on Jetpack 4.5. I didn't have time to work on it, so it was rightly closed for housekeeping reasons. 

It's a hair away from mergeable, but I just don't have the time to work on it right now. If other folks wanted to use it as a starting point for fixing Jetson compatibility they're more than welcome to.