TheSpaghettiDetective/obico-server #564: TSD local installation: ssl problem, need help with custom port probably?

pejotigrek, 2022-01-19:
long story short - I've got a PC with latest debian, and port 443 is already used by other installed software [HomeAssistant, if it's important]. public domain & SSL certificates configured with DuckDNS/Let's Encrypt [thanks to HomeAssistant's addons].
but still - I can't get SSL to work in TheSpaghettiDetective.

everytime I tried to set `SITE_USES_HTTPS: 'True'`, TSD dashboard and Django site stops working.

I'm starting to thing that maybe I should force TSD to use other port than 443 for https, I can make some forwards on the router-side, but Is there any way to configure TSD [via docker-compose maybe?] to listen to other port when using https?

encetamasb, 2022-01-24:
@pejotigrek Hi! Could you paste the output of ```curl -v https://<your site>``` here when ssl is enabled?

pejotigrek, 2022-01-25:
@encetamasb hello! sure, here is the response [I've hashed out the domain and IP addresses]:

```
########@cHAOS:~$ curl -v https://########.duckdns.org

*   Trying ########:443...
* Connected to ########.duckdns.org (########) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (IN), TLS handshake, Server key exchange (12):
* TLSv1.2 (IN), TLS handshake, Server finished (14):
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):
* TLSv1.2 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.2 (OUT), TLS handshake, Finished (20):
* TLSv1.2 (IN), TLS handshake, Finished (20):
* SSL connection using TLSv1.2 / ECDHE-ECDSA-AES256-GCM-SHA384
* ALPN, server accepted to use h2
* Server certificate:
*  subject: CN=########.duckdns.org
*  start date: Dec 30 01:09:02 2021 GMT
*  expire date: Mar 30 01:09:01 2022 GMT
*  subjectAltName: host "########.duckdns.org" matched cert's "########.duckdns.org"
*  issuer: C=US; O=Let's Encrypt; CN=R3
*  SSL certificate verify ok.
* Using HTTP2, server supports multi-use
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
* Using Stream ID: 1 (easy handle 0x5620d4313270)
> GET / HTTP/2
> Host: ########.duckdns.org
> user-agent: curl/7.74.0
> accept: */*
>
* Connection state changed (MAX_CONCURRENT_STREAMS == 128)!
< HTTP/2 200
< server: nginx
< date: Tue, 25 Jan 2022 07:50:18 GMT
< content-type: text/html; charset=utf-8
< content-length: 9315
< strict-transport-security: max-age=31536000; includeSubDomains
```

next there was just a html code that displays my HomeAssistant's dashboard.

encetamasb, 2022-01-25:
Thanks! This is probably a configuration issue. There is a #private-server-help channel on Discord, where you may find (better) help faster.

Without knowing more, I guess you have to set up a new *server* section in your nginx config and add a proper proxy_pass configuration. There you can set another, ssl enabled port to listen on; make sure to open that port in your router, and that related proxy_pass forwards requests to port 3334 (of TSD). You don't need to add or change ports in docker-compose.yml.

You can find an example nginx config here, what you need to adapt:

https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/docs/advanced_config.md#nginx

pejotigrek, 2022-01-25:
thank you @encetamasb for your answer!
the problem is, that I probably can't use given nginx config.. as it also is via the HomeAssistant's addon, and AFAIK there's no access to this level of configuration [it's mostly "click and forget" type of scenario]. but knowing that it is not a TSD issue, I'll try to channel my problem towards HA+nginx combo.

I'll keep this issue opened for a while though, and update it if I'll find the solution :)

encetamasb, 2022-01-25:
https://community.home-assistant.io/t/home-assistant-community-add-on-nginx-proxy-manager/111830 may help

pejotigrek, 2022-01-25:
> https://community.home-assistant.io/t/home-assistant-community-add-on-nginx-proxy-manager/111830 may help

thanks! I'll check that soon.
in the meantime I've got to know the way of adding "normal" config file into the addon, so I guess there's a light in the tunnel.. ;)

closed 2022-04-26 by raymondh2