TheSpaghettiDetective/obico-server #565: Manual Installation

damda58, 2022-01-20:
Hello,

I have a Proxmox server with several LXC containers. I wanted to know if it was possible to do a manual installation without going through docker?

encetamasb, 2022-01-24:
@damda58 It's possible, but there is no official documentation, and needs some effort. You need to recreate the environment for every service, just follow "base" and "normal" Dockerfile-s and the main docker-compose.yml file. 

It's possible that you need to fix hardcoded configuration in the code as well (maybe or maybe not). Anyway, if you did this and found issues, please report back, so we can fix them.

closed 2022-04-26 by raymondh2

livinhack, 2025-02-02:
It would really help if there was a docker-free installation script to install Obico Server natively on Linux. You could first set up an Ubuntu LXC in Proxmox, for example, and then trigger the installation via the console. Docker in the Lxc is a nightmare, it uses too many resources!