TheSpaghettiDetective/obico-server #570: Recognize stable upgrades

querex007, 2022-02-04:
I'm running TheSpaghettiDetective on my own docker.
For an 'upgrade', the documentation instructions say I need to fetch the master branch once again, build everything, run the dockers again.

Because there is no officiel release in github available.
Just wondering how I can recognize stable upgrades as I fetch the master branch.
Or is the master branch always the stable release branch?

Many thanks for the advice.
Keep up the awesome project!

encetamasb, 2022-02-04:
Hi @querex007, this project does not have the concept of stable releases. The master branch is as stable as possible, there is a chance that when you pull you get both improvements and fixes, and new (not yet found) bugs. I think, this situation is not great, but I know why it is managed that way (effort vs gain, from team perspective).

If you have any proposal, what would you expect and how would you do it, please share! Tagging "stable" versions and backporting fixes from master, that won't happen probably... but it's not impossible to convince the dev team (in general), as far as you have a good argument ;)

kennethjiang, 2022-02-05:
We have been following the best practices of Continuous Delivery and using the master branch as the stable and releasable branch to minimize the process overhead. However, I did realize we have the flexibility of rolling back TSD cloud easily in case of a broken master branch, whereas rollback is a lot trickier to perform for users who are running their TSD private server.

I just changed our process to using the `release` branch for the private server, while the TSD cloud stays on the `master` branch for maximum agility.

closed 2022-02-05 by kennethjiang

querex007, 2022-02-11:
Awesome, the release branch is a perfect fit for my needs! Thanks.

commit b2a0d552 (2022-02-05): Close #570.
commit 3b9f6f35 (2022-02-18): Close #570.