TheSpaghettiDetective/obico-server #581: TLS Connection with own PKI

jonwilliams84, 2022-03-01:
Hi, I run TSD in my K8s cluster that is exposed via Traefik Ingress and rewrites all routers to HTTPS.  I use my own PKI provided by Certmanager and Vault Provider. 3 Tier cert, CA, ICA and Server cert.

I have rolled out the CA bundle to all my machines via ansible and have verified that its trusted via a simple curl:

```
 curl -v https://tsd.k8s.home
*   Trying 10.0.8.1...
* TCP_NODELAY set
* Expire in 200 ms for 4 (transfer 0x1d57880)
* Connected to tsd.k8s.home (10.0.8.1) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: none
  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
* TLSv1.3 (IN), TLS handshake, Finished (20):
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.3 (OUT), TLS handshake, Finished (20):
* SSL connection using TLSv1.3 / TLS_AES_128_GCM_SHA256
* ALPN, server accepted to use h2
* Server certificate:
*  subject: CN=tsd.k8s.home
*  start date: Feb 23 14:32:02 2022 GMT
*  expire date: Mar 25 14:32:32 2022 GMT
*  subjectAltName: host "tsd.k8s.home" matched cert's "tsd.k8s.home"
*  issuer: CN=k8s.home Intermediate Authority
*  SSL certificate verify ok.
* Using HTTP2, server supports multi-use
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
* Using Stream ID: 1 (easy handle 0x1d57880)
> GET / HTTP/2
> Host: tsd.k8s.home
> User-Agent: curl/7.64.0
> Accept: */*
>
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* Connection state changed (MAX_CONCURRENT_STREAMS == 250)!
< HTTP/2 302
< content-type: text/html; charset=utf-8
< location: /accounts/login/
< vary: Origin, Cookie
< x-frame-options: SAMEORIGIN
< content-length: 0
< date: Tue, 01 Mar 2022 11:02:53 GMT
<
* Connection #0 to host tsd.k8s.home left intact
```

However, when I try and add my printer to TSD via Octoprint, it fails to connect. Checked the logs and this appears to be due to the CA Trust:

```
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/home/pi/oprint/lib/python3.7/site-packages/urllib3/connectionpool.py", line 756, in urlopen
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "/home/pi/oprint/lib/python3.7/site-packages/urllib3/util/retry.py", line 574, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='tsd.k8s.home', port=443): Max retries exceeded with url: /api/v1/octo/unlinked/ (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1056)')))
```

Is there anywhere else I need to add my CA bundle in order for Octoprint to trust it? Apologies if this is more a problem on the Octoprint side, but they do not monitor their issues seemingly.

Regards,

Jon

encetamasb, 2022-03-02:
Hi!

I don't think you can solve this with additional configuration. If you are ready to update the code, then as a first step look at this

https://stackoverflow.com/questions/10667960/python-requests-throwing-sslerror

and then set ```verify``` param here

https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/blob/master/octoprint_thespaghettidetective/utils.py#L272

This might not solve all issues.. but I assume it solves most.

kennethjiang, 2022-03-02:
Closing it as this is not a code issue. The quickest way to get help on an environment issue like this is our discord server: https://discord.gg/NcZkQfj @jonwilliams84

closed 2022-03-02 by kennethjiang